RobotLocomotion/drake #21139: Meshcat animations

SeanCurtis-TRI, 2024-03-13:
### What happened?

During the effort of implementing a solution to #19142, we became aware of a number of quirks with the animation mechanism.

The goal of this issue is as follows:
  1. Enumerate the current behavior (at the time the issue was authored).
  2. Determine if that behavior is intended/acceptable and if our documentation is sufficient to the behavior.
  3. Determine if there's any future work we need to do to change the behavior/documentation.

## Current behavior

TBD (Ran out of time today, I'll come back to fill this in later. This is a concrete placeholder to remind myself to capture this. Harass me if you don't see it.)

jwnimmer-tri, 2024-05-01:
> Harass me if you don't see it.

Ping.

jwnimmer-tri, 2024-05-28:
Per f2f: closing as currently unactionable.

closed 2024-05-28 by jwnimmer-tri