RobotLocomotion/drake #21140: Remove mac x86 vSphere from Jenkins

svenevs, 2024-03-14:
Relates: #19057

This ticket is tracking removing mac x86 via vSphere from jenkins.  The changes are ready in https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/124 , more information there.

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2024-03-21:
For clarity -- must NOT happen prior to mid-April release.  It should be OK to do anytime after April release, but before the x86 machines die.

jwnimmer-tri, 2024-04-18:
We're ready for Phase 3 now (and must happen in the next <9 days).

jwnimmer-tri, 2024-04-22:
Remember two related points:
- Adjust [release_playbook]https://drake.mit.edu/release_playbook.html] to remove the mention of the staging job and the x86 wheels in the tallies.
- Adjust [download_release_candidate.py](https://github.com/RobotLocomotion/drake/blob/master/tools/release_engineering/download_release_candidate.py) to remove the x86 downloading (no more staging).

For the playbook, the new text will be something at the end saying "File a ticket against Jeremy to manually release the x86 wheels" but I'll PR that separately.

BetsyMcPhail, 2024-04-23:
All Mac Monterey VSphere jobs have been removed from Jenkins.

BetsyMcPhail, 2024-04-23:
Before closing this issue, there is also some cleanup to do in drake-ci

jwnimmer-tri, 2024-04-23:
Some docs still mention these jobs, which will also need some cleanups:
- https://drake.mit.edu/jenkins.html
- https://drake.mit.edu/release_playbook.html
- ... maybe more?  I didn't grep.  Those were what I know off the top of my head.

PR #21353 (2024-04-24): Remove mac x86 wheel jobs
Resolves #21140

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21353)
closed 2024-04-24 by jwnimmer-tri