RobotLocomotion/drake #21154: pydrake binding of `ContextBase::GetSystemPathname`

ggould-tri, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
I am writing a module depending on pydrake and want to ensure that my contexts and systems match and debug a mismatch.
I want to write and test the python equivalent of the C++ condition that's failing,
```
  DRAKE_DEMAND(plant_context.GetSystemPathname() ==
               plant.GetSystemPathname());
```

**Describe the solution you'd like**
pydrake contexts should bind `GetSystemPathname`

**Describe alternatives you've considered**
Searched for https://drake.mit.edu/pydrake/search.html?q=pathname&check_keywords=yes&area=default and didn't find anything helpful.

jwnimmer-tri, 2024-03-15:
Certainly all public API should be bound, so I have no objection to the bindings request.

However, there's one thing worth clarifying: the canonical way to check that a Context is appropriate to use with a System is `SystemBase::ValidateContext()`, not by checking anything to do with `GetSystemPathname()`.  The C++ code's formulation of the check is probably wrong.

The `system.ValidateContext(context)` method is already bound in pydrake.

jwnimmer-tri, 2024-03-19:
In general we don't keep issues open for "create a binding for Foo".  When someone needs it, they just PR it.  Or sometimes, we'll go through and backfill all missing bindings for a given package, all at once.

Given the non-essential priority, I'll just close.

closed 2024-03-19 by jwnimmer-tri