RobotLocomotion/drake #21168: Sanitizer timeouts on //planning/trajectory_optimization:direct_transcription_test

svenevs, 2024-03-19:
Today marks the second occurrence of this.  [First occurrence](https://drakedevelopers.slack.com/archives/C270MN28G/p1675979066422879) was on February 9th 2023 (`linux-focal-clang-bazel-continuous-everything-leak-sanitizer`).

It's not clear if the changes in https://github.com/RobotLocomotion/drake/pull/21138 would somehow change the plumbing enough to make https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/387/console timeout.

For now just posting the issue to make it searchable, status quo: relaunch and ignore.

xuchenhan-tri, 2024-03-19:
I'd be surprised if #21138 has anything to do with this particular timeout. The heavylifting in the test should happen inside solving the optimization problem with SNOPT.

xuchenhan-tri, 2024-03-28:
This happened again in [linux-jammy-clang-bazel-continuous-everything-address-sanitizer/424](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/424)

xuchenhan-tri, 2024-06-25:
Again on https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-coverage/262/

SeanCurtis-TRI, 2025-01-06:
Two more instances:

2025-01-02 - https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-everything-leak-sanitizer/426/
2025-01-06 - https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-everything-leak-sanitizer/430/

For the record, a quick, informal perusal of the previous few weeks of tests show that the test runs close to the time limit (55-58 s would be considered a typical time for a successful test).

BetsyMcPhail, 2025-01-13:
1/13 https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug/929/

BetsyMcPhail, 2025-01-17:
1/17 - [linux-jammy-clang-bazel-nightly-everything-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-everything-leak-sanitizer/444/)

BetsyMcPhail, 2025-02-12:
2/12  https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug/960/

BetsyMcPhail, 2025-02-14:
2/14 - https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-everything-debug/361/

tyler-yankee, 2025-02-19:
2/19: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-leak-sanitizer/478/

tyler-yankee, 2025-02-20:
2/20: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug/968/

tyler-yankee, 2025-03-04:
3/4: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-leak-sanitizer/493/

BetsyMcPhail, 2025-04-09:
4/9 http://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-everything-debug/414/

Aiden2244, 2025-04-17:
4/17
- https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug/1026/
- https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-everything-debug/422/

Aiden2244, 2025-04-18:
4/18 https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-everything-debug/423/changes

ggould-tri, 2025-05-07:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-gcc-bazel-nightly-everything-debug/443/

Aiden2244, 2025-06-23:
6/23 https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-everything-debug/491

BetsyMcPhail, 2025-07-11:
7/11 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-everything-debug/509/

xuchen-han, 2025-08-04:
8/4 https://drake-jenkins.csail.mit.edu/view/Production/job/linux-noble-unprovisioned-gcc-bazel-nightly-everything-debug/48/