RobotLocomotion/drake #21171: Make `CalcRobotClearance()` return k-closest signed-distance pairs

siddancha, 2024-03-20:
**The problem**
I'm using `SceneGraphCollisionChecker::CalcRobotClearance()` to find the signed distance between the robot and the world, as well as the gradient of signed distance. The issue is that I only really care about the smallest signed-distance pair (and its derivative). So I loop through the results and retrieve the minimal distance pair. However, this is slow because [CalcRobotClearance()](https://github.com/RobotLocomotion/drake/blob/deb04db6a1eedd009b1c51555d09d71234bd5671/planning/scene_graph_collision_checker.cc#L197-L295) loops over all signed-distance pairs and calls the expensive `CalcJacobianPositionVector()` for each pair. I only end up using one of these results, among potentially thousands of collision geometry pairs.

**Proposed solution**
A solution could be to add a parameter `num_closest_pairs` to `CalcRobotClearance()` that returns that many closest pairs in `RobotClearance`. This can be efficiently implemented by first filtering out the `num_closest_pairs` number of pairs, and calling `CalcJacobianPositionVector()` only on those pairs. I intend to set `num_closest_pairs=1`, but other users may require more such pairs. The default value of `num_closest_pairs` could be `std::nullopt` that exhibits the original behavior of returning all pairs.

RussTedrake, 2024-03-21:
+@SeanCurtis-tri -- can you help with disposition as @hongkai-dai is out and I believe you understand this well.
cc @calderpg-tri ... would love to hear your thoughts.

SeanCurtis-TRI, 2024-03-21:
Several thoughts:

1. The suggested approach is fine. At least for `SceneGraphCollisionChecker`.
2. I'd like to hear from @calderpg-tri to see if he believes that kind of argument would be difficult for some of the other types of `CollisionChecker` that still live in anzu. My gut says it shouldn't be a problem.
3. A further optimization would be to push the `num_closest_pairs` down into the `QueryObject` API -- it'd be nice if we weren't continuing evaluating contact for pairs that we know must be farther than the first N.

jwnimmer-tri, 2024-03-21:
> A further optimization would be to push the num_closest_pairs down into the QueryObject API -- it'd be nice if we weren't continuing evaluating contact for pairs that we know must be farther than the first N.

A fine idea on its own merits, but I don't immediately see how it would help here.  The collision checker has padding that will affect which K are closest, so I expect we probably need all of the distances from SG.

SeanCurtis-TRI, 2024-03-21:
Excellent point. I'd forgotten about the pairwise padding.

There's a part of my soul that believes there could be an appropriate affordance for the padding, but I'll have to ponder that.

calderpg-tri, 2024-03-21:
If I understand your current use case, you do not actually want `k=1`-closest of _all_ pairs, you want the `k=1`-closest _robot<->environment_ pair. Those are actually quite different requests, since for any given configuration there could be arbitrarily many robot<->robot pairs closest than the closest robot<->environment pair.

I do think that closest-pair only queries are useful enough to get a fast path, but I do not think complicating the `CalcRobotClearance` API and implementations to express them is the right approach. What I would propose instead is a new method `CalcClosestRobotClearance` which returns a `RobotClearance` with 0, 1, or 2 entries:

- 2 entries: if the model contains both robot<->robot and robot<->environment distances within influence distance threshold, one entry for the closest robot<->robot pair and one entry for the closest robot<->environment pair.
- 1 entry: if the model contains either robot<->robot or robot<->environment distance within influence distance threshold, one entry for either the closest robot<->robot or the closest robot<->environment distance pair.
- 0 entries: if the model contains no distances within influence distance threshold, no entries are returned.

With that API you always get the closest robot<->robot and robot<->environment pairs (if they exist), regardless of where they would be in a combined ordering by distance. Potentially that can be expanded to support the `k`-closest, but the simplest implementation and most consistent cross-`CollisionChecker`-implementation behavior exists for the `k=1` case.

siddancha, 2024-03-21:
> If I understand your current use case, you do not actually want k=1-closest of all pairs, you want the k=1-closest robot<->environment pair. Those are actually quite different requests, since for [...]

@calderpg-tri I am adding collision filters between pairs of robot geometries that I want to exclude. So my use case is: return at most `k=1` closest collision pairs between geometries that `CalcRobotClearance()` would have originally computed taking into account collision filters. I think shifting the burden on the user to specify collision filters might make a cleaner API for `CalcRobotClearance()` or the proposed `CalcClosestRobotClearance()`.

I would have a slight preference for an expanded `CalcClosestRobotClearance` to support the `k`-closest, since I might take a weighted average >1 collision pairs to avoid jerky controller behavior when the closet pair changes. But I do agree that `CalcClosestRobotClearance` for `k=1` is cleaner and I can work with that.

calderpg-tri, 2024-03-21:
> I think shifting the burden on the user to specify collision filters might make a cleaner API for CalcRobotClearance() or the proposed CalcClosestRobotClearance().

I do not think this is a reasonable burden to impose on the user. In the proposed `CalcClosestRobotClearance`, if the user only wants robot<->robot or robot<->environment distances, they only need to ignore part of the returned `RobotClearance`. If they know they will only _ever_ want either robot<->robot or robot<->environment distances, they can set filters as an additional optimization. Otherwise, if we only returned the single closest distance, a user wanting both the closest robot<->robot and robot<->environment distances would have to continuously swap filters. Two queries, even with much of the geometry filtered, are almost certainly not going to be cheaper than one query (filtered geometries are cheaper, but not free), and the process of applying filters is not free either.

> I would have a slight preference for an expanded CalcClosestRobotClearance() to support the k-closest, since I might need a weighted average over k>1 collision pairs to avoid jerky controller behavior when the closest pair changes.

This is not possible to achieve in the general case. Imagine a robot near the middle of a room - if you move in any direction, the `k` nearest distances can change from being the distances to objects on one wall to the distances to objects on another. Selecting `k` in complicated environments to any value other than "all" becomes difficult; this is why we provide `influence_distance` instead to set a bound, since it is generally much clearer what a distance bound means than a bound on number of distance pairs.

siddancha, 2024-03-21:
> I do not think this is a reasonable burden to impose on the user. In the proposed `CalcClosestRobotClearance`, if the user only wants robot<->robot or robot<->environment distances, they only need [...]

I see your point, agreed! I didn't consider the use-case when the user would want both robot↔robot and robot↔environment distances.

> This is not possible to achieve in the general case. Imagine a robot near the middle of a room - if you move in any direction, the k nearest distances can change from being the distances to objects [...]

Makes sense; `k=1` suffices for me.