RobotLocomotion/drake #21176: zip: command not found in test setup script.

xuchenhan-tri, 2024-03-21:
There are now two instances of CI test failure with the following message:

In [linux-jammy-gcc-bazel-weekly-snopt-coverage/25](https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-jammy-gcc-bazel-weekly-snopt-coverage/25/):

```
[7:45:09 AM]  ==================== Test output for //manipulation/util:move_ik_demo_base_test:
[7:45:09 AM]  -- Test timed out at 2024-03-17 14:45:08 UTC --
[7:45:09 AM]  external/bazel_tools/tools/test/test-setup.sh: line 428: zip: command not found
[7:45:09 AM]  Could not create "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-weekly-snopt-coverage/_bazel_ubuntu/ac163a28ce4d58af9323a9acdac38c75/execroot/drake/bazel-out/k8-dbg/testlogs/manipulation/util/move_ik_demo_base_test/test.outputs/outputs.zip": 
```

Then in [linux-jammy-gcc-bazel-nightly-coverage/163](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-coverage/163/)

```
[2:05:17 AM]  /media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/_bazel_ubuntu/13ab93db129735c0cc7824860397095a/execroot/drake/bazel-out/k8-dbg/bin/systems/framework/supervector_test.runfiles/drake/tools/dynamic_analysis/kcov: line 65: 156087 Segmentation fault      (core dumped) kcov "--include-path=/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src" --verify --python-parse=python3 --exclude-pattern=third_party "--replace-src-path=/proc/self/cwd:/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src" "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/_bazel_ubuntu/13ab93db129735c0cc7824860397095a/execroot/drake/bazel-out/k8-dbg/testlogs/systems/framework/supervector_test/test.outputs/kcov" systems/framework/supervector_test
[2:05:17 AM]  external/bazel_tools/tools/test/test-setup.sh: line 428: zip: command not found
[2:05:17 AM]  Could not create "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/_bazel_ubuntu/13ab93db129735c0cc7824860397095a/execroot/drake/bazel-out/k8-dbg/testlogs/systems/framework/supervector_test/test.outputs/outputs.zip": 
```

In each case, (as far as I can tell), the `zip: command not found` is triggered by an actual cause of the test failure (timeout in the first example and segfault in kcov in the second example). However, I can't be sure of the root cause of the failure, hence documenting this in an issue per buildcop playbook.

closed 2024-03-21 by xuchenhan-tri

jwnimmer-tri, 2024-03-21:
https://github.com/bazelbuild/bazel/issues/13230 seems related.

I guess the `bazel.deb` fails to declare `Depends: zip` so it's possibly not installed CI.

I'm unclear on why this only bites us on coverage timeouts, though.  I would have thought all kcov builds would be sad about it.

BetsyMcPhail, 2024-04-01:
4/1: [linux-jammy-gcc-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-coverage/176/)

jwnimmer-tri, 2024-04-01:
For the record, the "zip: command not found" is printed on _any_ kind of failure during a coverage test, so isn't a useful collation mechanism.

The relevant failures so far have been either "segmentation fault" or "timeout".