RobotLocomotion/drake #21192: Error in query_object.ComputeSignedDistancePairClosestPoints

cohnt, 2024-03-26:
### What happened?

I've encountered a problem where calling `query_object.ComputeSignedDistancePairClosestPoints(geomA, geomB)` yields the error `Failure at geometry/proximity/distance_to_shape_touching.cc:73 in PointOnBoxSurfaceHelper(): condition 'n != Vector3d::Zero()' failed.`.

A standalone reproduction of this bug can be found in [this gist](https://gist.github.com/cohnt/f720c21caf07f7daf3313724e2db1676).

cc @DamrongGuoy

DamrongGuoy, 2024-03-27:
I can reproduce the problem in a temporary [notebook](https://github.com/DamrongGuoy/drake/blob/bugfix_issue21192/tutorials/debug_ComputeSignedDistancePair.ipynb) based on your gist.  Thank you very much.  It also gave me some ideas to fix it.

Next, I'll create a unit test and find a good solution.

DamrongGuoy, 2024-03-27:
I have created a unit test in the second [commit](https://github.com/RobotLocomotion/drake/commit/f93a0d9732eb4be1fb1a12d23cb4e32c41ca9871) in my branch.

I'm working towards a solution.

DamrongGuoy, 2024-04-03:
For later discussion, the two boxes (geomA and geomB) are shown in this picture.  This is my WIP [branch](https://github.com/DamrongGuoy/drake/tree/bugfix_issue21192).

[image]

DamrongGuoy, 2024-04-03:
@cohnt would it be easy for you to try my WIP branch https://github.com/DamrongGuoy/drake/tree/bugfix_issue21192 with your work for [IrisInConfigurationSpace](https://drake.mit.edu/doxygen_cxx/group__geometry__optimization.html#ga3a51e0fec449a0abcf498f78a2a390a8) ?  I'd like to see whether there are more challenging cases before creating a PR.

Thank you for providing data.

cohnt, 2024-04-03:
Using the branch solves this issue, but I still encounter the problem described in #21234. Since my current workaround is just to wrap the signed distance query in a try-catch block, it's hard to know what is erroring (if it's erroring at all).

PR #21266 (2024-04-18): [geometry] Fix gradient of signed distance between touching (zero distance) boxes
Fix #21192

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21266)
closed 2024-04-18 by DamrongGuoy