RobotLocomotion/drake #21195: Build errors upon trying to import `multibody_plant.h`

alberthli, 2024-03-26:
### What happened?

I'm currently using `drake` in a `docker` container running `ubuntu` 22.04 installed by following [these instructions](https://drake.mit.edu/apt.html#stable-releases). In a test `c++` project that depends on `drake`, upon trying to import `multibody_plant.h`, I see the below error. This error only occurred today when I rebuilt my container without the Docker cache without any modification to the lines regarding `drake`, so it seems to be related to one of the newer releases, though I can't say which one with certainty. I do think the last time I built from scratch, the most recent `drake` release was 1.26.0.
```
1.867 In file included from /opt/drake/include/drake/geometry/collision_filter_declaration.h:7,
1.867                  from /opt/drake/include/drake/geometry/collision_filter_manager.h:4,
1.867                  from /opt/drake/include/drake/geometry/scene_graph.h:9,
1.867                  from /opt/drake/include/drake/multibody/plant/multibody_plant.h:21,
1.867                  from /home/test/test.cpp:1:
1.867 /opt/drake/include/drake/geometry/geometry_set.h: In member function 'bool drake::geometry::GeometrySet::contains(drake::geometry::FrameId) const':
1.867 /opt/drake/include/drake/geometry/geometry_set.h:285:23: error: 'const class std::unordered_set<drake::Identifier<drake::geometry::FrameTag> >' has no member named 'contains'
1.867   285 |     return frame_ids_.contains(frame_id);
1.867       |                       ^~~~~~~~
1.867 /opt/drake/include/drake/geometry/geometry_set.h: In member function 'bool drake::geometry::GeometrySet::contains(drake::geometry::GeometryId) const':
1.867 /opt/drake/include/drake/geometry/geometry_set.h:291:26: error: 'const class std::unordered_set<drake::geometry::GeometryId>' has no member named 'contains'
1.867   291 |     return geometry_ids_.contains(geometry_id);
1.867       |                          ^~~~~~~~
1.883 In file included from /opt/drake/include/drake/geometry/collision_filter_manager.h:6,
1.883                  from /opt/drake/include/drake/geometry/scene_graph.h:9,
1.883                  from /opt/drake/include/drake/multibody/plant/multibody_plant.h:21,
1.883                  from /home/test/test.cpp:1:
1.883 /opt/drake/include/drake/geometry/proximity/collision_filter.h: In member function 'bool drake::geometry::internal::CollisionFilter::HasGeometry(drake::geometry::GeometryId) const':
1.883 /opt/drake/include/drake/geometry/proximity/collision_filter.h:129:64: error: 'const FilterState' {aka 'const class std::unordered_map<drake::geometry::GeometryId, std::unordered_map<drake::geometry::GeometryId, drake::geometry::internal::CollisionFilter::PairRelationship> >'} has no member named 'contains'
1.883   129 |   bool HasGeometry(GeometryId id) const { return filter_state_.contains(id); }
1.883       |                                                                ^~~~~~~~
2.111 In file included from /opt/drake/include/drake/geometry/query_object.h:9,
2.111                  from /opt/drake/include/drake/geometry/scene_graph.h:13,
2.111                  from /opt/drake/include/drake/multibody/plant/multibody_plant.h:21,
2.111                  from /home/test/test.cpp:1:
2.111 /opt/drake/include/drake/geometry/query_results/deformable_contact.h: In member function 'bool drake::geometry::internal::DeformableContact<T>::IsRegistered(drake::geometry::GeometryId) const':
2.111 /opt/drake/include/drake/geometry/query_results/deformable_contact.h:383:36: error: 'const class std::unordered_map<drake::geometry::GeometryId, drake::geometry::internal::ContactParticipation>' has no member named 'contains'
2.111   383 |     return contact_participations_.contains(id);
2.111       |                                    ^~~~~~~~
2.177 In file included from /opt/drake/include/drake/geometry/geometry_roles.h:9,
2.177                  from /opt/drake/include/drake/geometry/render/render_engine.h:14,
2.177                  from /opt/drake/include/drake/geometry/query_object.h:14,
2.177                  from /opt/drake/include/drake/geometry/scene_graph.h:13,
2.177                  from /opt/drake/include/drake/multibody/plant/multibody_plant.h:21,
2.177                  from /home/test/test.cpp:1:
2.177 /opt/drake/include/drake/geometry/geometry_properties.h: In member function 'bool drake::geometry::GeometryProperties::HasGroup(const string&) const':
2.177 /opt/drake/include/drake/geometry/geometry_properties.h:236:20: error: 'const class std::unordered_map<std::__cxx11::basic_string<char>, std::unordered_map<std::__cxx11::basic_string<char>, drake::copyable_unique_ptr<drake::AbstractValue> > >' has no member named 'contains'
2.177   236 |     return values_.contains(group_name);
2.177       |                    ^~~~~~~~
2.301 In file included from /opt/drake/include/drake/geometry/scene_graph_inspector.h:17,
2.301                  from /opt/drake/include/drake/geometry/query_object.h:15,
2.301                  from /opt/drake/include/drake/geometry/scene_graph.h:13,
2.301                  from /opt/drake/include/drake/multibody/plant/multibody_plant.h:21,
2.301                  from /home/test/test.cpp:1:
2.301 /opt/drake/include/drake/geometry/internal_frame.h: In member function 'bool drake::geometry::internal::InternalFrame::has_child(drake::geometry::FrameId) const':
2.301 /opt/drake/include/drake/geometry/internal_frame.h:115:26: error: 'const class std::unordered_set<drake::Identifier<drake::geometry::FrameTag> >' has no member named 'contains'
2.301   115 |     return child_frames_.contains(frame_id);
2.301       |                          ^~~~~~~~
2.301 /opt/drake/include/drake/geometry/internal_frame.h: In member function 'bool drake::geometry::internal::InternalFrame::has_child(drake::geometry::GeometryId) const':
2.301 /opt/drake/include/drake/geometry/internal_frame.h:125:30: error: 'const class std::unordered_set<drake::geometry::GeometryId>' has no member named 'contains'
2.301   125 |     return child_geometries_.contains(geometry_id);
2.301       |                              ^~~~~~~~
2.302 In file included from /opt/drake/include/drake/common/eigen_types.h:20,
2.302                  from /opt/drake/include/drake/common/eigen_autodiff_types.h:17,
2.302                  from /opt/drake/include/drake/common/autodiff.h:27,
2.302                  from /opt/drake/include/drake/common/default_scalars.h:3,
2.302                  from /opt/drake/include/drake/multibody/plant/multibody_plant.h:17,
2.302                  from /home/test/test.cpp:1:
2.302 /opt/drake/include/drake/geometry/internal_frame.h: In member function 'void drake::geometry::internal::InternalFrame::remove_child(drake::geometry::GeometryId)':
2.302 /opt/drake/include/drake/geometry/internal_frame.h:138:5: error: 'class std::unordered_set<drake::geometry::GeometryId>' has no member named 'contains'
2.302   138 |     DRAKE_ASSERT(child_geometries_.contains(geometry_id));
2.302       |     ^~~~~~~~~~~~
2.302 /opt/drake/include/drake/geometry/internal_frame.h:138:5: error: 'class std::unordered_set<drake::geometry::GeometryId>' has no member named 'contains'
2.302 /opt/drake/include/drake/geometry/internal_frame.h:138:5: error: template argument 1 is invalid
2.302   138 |     DRAKE_ASSERT(child_geometries_.contains(geometry_id));
2.302       |     ^~~~~~~~~~~~
[... 229 lines cut ...]
find_package(drake REQUIRED)
add_executable(test test.cpp)
target_link_libraries(test drake::drake)
```

The file `test.cpp`:
```
#include <drake/multibody/plant/multibody_plant.h>

int main(int argc, char** argv) {
    return 0;
}
```

### Version

1.27.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

apt install drake

jwnimmer-tri, 2024-03-26:
See https://drake.mit.edu/release_notes/v1.27.0.html:

> As a result, we have some new lower bounds on supported tool versions:
> C++ standard >= 20

You need to tell CMake to use C++20, like [this](
https://github.com/RobotLocomotion/drake-external-examples/commit/75c2cc18c7f8a6218db2031ac8cd195dee2623b2#diff-204d768701f990d4f53ac96479b9cd87318a705e905d525630a6fcc7e824de98).

closed 2024-03-26 by jwnimmer-tri

jwnimmer-tri, 2024-04-02:
We found some places that still said C++17, so possibly as of #21247 this would be automatically fixed.