RobotLocomotion/drake #21215: SceneGraphCollisionChecker throws when models have a QuaternionFloatingJoint (for a non-floating body)

RussTedrake, 2024-03-29:
### What happened?

This was originally reported in https://stackoverflow.com/questions/78208761/drake-faking-attachment-between-floating-objects-with-locking-joints-is-it-po .

I've made a c++ reproduction that can be dropped in to `scene_graph_collision_checker_test.cc`:
```
// Reproduction of issue from
// https://stackoverflow.com/questions/78208761/drake-faking-attachment-between-floating-objects-with-locking-joints-is-it-po
GTEST_TEST(SceneGraphCollisionCheckerTest, QuaternionFloatingJoint) {
  RobotDiagramBuilder<double> builder;
  const std::string model_directives = R"""(
directives:
- add_model:
    name: arm
    file: package://drake/manipulation/models/iiwa_description/urdf/iiwa14_spheres_dense_collision.urdf
- add_weld:
    parent: world
    child: arm::base
- add_model:
    name: cracker
    file: package://drake_models/ycb/003_cracker_box.sdf
)""";
  builder.parser().AddModelsFromString(model_directives, "dmd.yaml");
  
  auto joint = std::make_unique<multibody::QuaternionFloatingJoint<double>>(
    "lockable_floating_joint",
    builder.plant().GetFrameByName("iiwa_link_ee"),
    builder.plant().GetFrameByName("base_link_cracker"));
  builder.plant().AddJoint(std::move(joint));

  const auto& plant = builder.plant();
  CollisionCheckerParams params;
  params.model = builder.Build();
  params.robot_model_instances.push_back(plant.GetModelInstanceByName("arm"));
  params.edge_step_size = 0.05;
  SceneGraphCollisionChecker dut(std::move(params));

  // As constructed, collision filters must be consistent.
  EXPECT_NO_THROW(EnforceCollisionFilterConsistency(dut));
}
```
(you'll also need to `#include "drake/multibody/tree/quaternion_floating_joint.h"` at the top)

The result is 
```
abort: Failure at bazel-out/k8-dbg/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/rigid_body.h:322 in floating_positions_start(): condition 'is_floating()' failed.
```

Running it through the debugger, the offending code is [here](https://github.com/RobotLocomotion/drake/blob/e41e11f560650c3b9ba805d4587adb6c42dc7b8b/planning/linear_distance_and_interpolation_provider.cc#L32).

I'll assign to @SeanCurtis-TRI , since his comments are all over this.  cc @joemasterjohn (since it's related to quaternionfloatingjoint / floating joints) and @calderpg-tri (for collision checker).

### Version

master

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
abort: Failure at bazel-out/k8-dbg/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/rigid_body.h:322 in floating_positions_start(): condition 'is_floating()' failed.
```
```

jwnimmer-tri, 2024-03-29:
My guess is that now that all dofs have joints, `GetQuaternionDofStartIndices` should loop over all joints instead of all bodies.

lcbw, 2024-04-15:
(I'm the Stack Overflow OP): Is there a workaround for this bug currently like leveraging a normal scene graph? Thank you!

jwnimmer-tri, 2024-04-15:
The only work-around I can think of is using a different SceneGraph for collision checking vs simulation, where the collision checking scene graph omits the lockable floating joint.

lcbw, 2024-04-15:
Hmm - It looks like this also impacts the ability to check the IK validity of start and end poses. 
ik = InverseKinematics(plant, plant_context)
SystemExit: Failure at bazel-out/k8-opt/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/rigid_body.h:322 in floating_positions_start(): condition 'is_floating()' failed.

closed 2024-04-17 by calderpg-tri

calderpg-tri, 2025-01-27:
> does https://github.com/RobotLocomotion/drake/pull/21308 also solve this issue for inverse kinematics?

No. Anywhere else that is looking for quaternion DoF via `floating_positions_start()` is still broken.

marip8, 2025-01-27:
Okay. I'll open a corresponding issue then

PR #21308 (2024-04-17): [planning] Fix GetQuaternionDofStartIndices to loop over joints, not bodies
I believe this resolves #21215, @SeanCurtis-TRI did you have something else in mind in your TODO?

+@SeanCurtis-TRI for feature review, thanks!

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21308)