RobotLocomotion/drake #21224: Cannot run the visualization tool about Meshcat on my own Machine! Can’t open the link “https://localhost:7000”

BowenFF, 2024-03-31:
### What happened?

When I use pydrake in my own machine and use the IDE like vscode.
I run the code about the manipulation lectures. 
And when I run “meshcat = Start Meshcat”, It returns the Information like”INFO:drake:Meshcat listening for connections at https://localhost:7000”. But when I use the Firefox browser to open the link,it shows that “Firefox cannot build to localhost:7000 serve connection “.
How can I do to solve this problem?
Configuration:Ubuntu20.04 Pydrake Anaconda’s Virtual environment.

### Version

Latest version

### What operating system are you using?

Other

### What installation option are you using?

pip install drake

jwnimmer-tri, 2024-04-02:
This bug report doesn't contain enough information for us to reproduce the problem, so there's not much we can do to help.

Therefore, I'll close it for now, but you can still reply with more information.  Closing just means it's not actionable, it doesn't prevent new comments.

At minimum, we would need:
- the exact commands you are running (not just "I run the code about the manipulation lectures"); and
- the exact copy-pasted error message(s).

> INFO:drake:Meshcat listening for connections at `https://localhost:7000`

Does it really say `https` (not `http`)?  I don't know where the `https` would be coming from, so that might be the nexus of your problem.

> Ubuntu 20.04

Drake no longer supports 20.04 (https://drake.mit.edu/release_notes/v1.27.0.html).

> Anaconda’s Virtual environment.

Drake does not support Anaconda (https://drake.mit.edu/pip.html#stable-releases).

closed 2024-04-02 by jwnimmer-tri

BowenFF, 2024-04-03:
**I know how to fix it.** I have used the VScode IDE to  run my code. If I run it directly , I can’t open the link:http:localhost:7000. However, if I use the function of VScode, which is “Run in interactive Window “->”Run current file in interactive Window “. This function just like the jupyter notebook . Then I find it works. Surprisingly! But I don’t know why? At the same time,I can monitor the port 7000.（sudo netstat -nlput|grep 7000）If you know the reason，please tell me. Thanks！

jwnimmer-tri, 2024-04-03:
I don't think we know.  The only VSCode-related information I have is https://stackoverflow.com/questions/77137333/meshcat-port-forwarding-with-vs-code which isn't much.