RobotLocomotion/drake #21225: Deprecation removal contact_solver

jwnimmer-tri, 2024-04-01:
The MbP has some deprecations related to the contact solver:

https://github.com/RobotLocomotion/drake/blob/f8f957f0817bcfc8d789452e1a80e6348d545f5f/multibody/plant/multibody_plant.h#L2235-L2238

https://github.com/RobotLocomotion/drake/blob/f8f957f0817bcfc8d789452e1a80e6348d545f5f/multibody/plant/multibody_plant_config.h#L64-L73

Normally I'd just remove the deprecated code myself, but I can't this time because I don't have the "information we collect from users".

Please remove the deprecated code / comments and adjust the nearby code as-necessary.

jwnimmer-tri, 2024-04-09:
I think probably this ticket is holding up #21095, so I hope to have us resolve this sooner than later.

@amcastro-tri As of today, which of "similar" or "lagged" (or "sap?) is our best guess at the most useful default value?  I can open the PR if you like, I just don't know which one to choose.

amcastro-tri, 2024-04-10:
Per f2f: running kLagged would be ideal in that it does not introduce artifacts results of the convexification (i.e. "gliding" when sliding). However, kLagged introduced a "weak" coupling of  friction and normal forces that might impact the perfomance (meaning predicted success/failure) of grasping tasks in simulation. In this regard, kSimilar is the safe bet, in that it solves most of the big issues we had with kSap and introduces Hunt & Crossley for the accurate and stable modeling of dissipation.

See #21294, thus far a WIP until we confirm all tests pass. I run a few tests locally and they succeded. Lets wait for Drake CI and Anzu CI for the final veredict.

jwnimmer-tri, 2024-11-01:
@amcastro-tri Would you like to add this to our project board, for yourself?

closed 2024-12-17 by jwnimmer-tri