RobotLocomotion/drake #21255: CI failure: "Missing digest"

BetsyMcPhail, 2024-04-03:
On 4/2/24, buildcops began noticing CI jobs failing with errors like:

`[10:15:14 AM]  ERROR: /media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-continuous-debug/src/math/BUILD.bazel:427:20: Compiling math/test/discrete_algebraic_riccati_equation_test.cc failed: unable to finalize action: Missing digest: 6f5ad5446e49451a55dddee73681d3f44103b4aea9c85b308a9761effe9e3693/34778 for bazel-out/k8-dbg/bin/math/_objs/discrete_algebraic_riccati_equation_test/discrete_algebraic_riccati_equation_test.pic.d`

The build cache key was incremented in https://github.com/RobotLocomotion/drake-ci/pull/276 as an attempted solution, but the failure continues to sporadically appear. 

See [slack conversation](https://drakedevelopers.slack.com/archives/C270MN28G/p1712073402141219) for initial discussion.

BetsyMcPhail, 2024-04-15:
4/15: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/495/

SeanCurtis-TRI, 2024-04-18:
4/18 

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/514/

xuchenhan-tri, 2024-04-24:
4/24
[https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/537/](https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/537/)

BetsyMcPhail, 2024-05-20:
5/20
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-debug/283/ and https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-debug/284/

mwoehlke-kitware, 2024-05-28:
5/28
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/622/

BetsyMcPhail, 2024-06-04:
6/4 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-debug/320/

xuchenhan-tri, 2024-06-25:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-clang-bazel-continuous-debug/351
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-clang-bazel-continuous-debug/352

BetsyMcPhail, 2024-08-06:
8/5 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/787/

mwoehlke-kitware, 2024-08-06:
8/6 https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/796/

BetsyMcPhail, 2024-09-10:
9/10
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/867/
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/868/

nicolecheetham, 2024-10-09:
10/9 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/960/

williamjallen, 2024-10-16:
10/16

https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/984/consoleFull

nicolecheetham, 2024-11-06:
11/6 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1041/consoleFull

BetsyMcPhail, 2024-12-12:
12/12 cont.

https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/2239/
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1067/
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-debug/745/

BetsyMcPhail, 2025-01-06:
1/6: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1140

ggould-tri, 2025-01-08:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-gcc-bazel-continuous-debug/2330/consoleFull again x4

jwnimmer-tri, 2025-01-08:
For the record, I believe there is not much we can do to fix this.  The rate is elevated this week because I'm pushing dozens of PRs that invalidate the entire build cache.  Once that quiets down, the error frequency should drop back down to a more nominal rate.

jwnimmer-tri, 2025-01-08:
FYI If it's a PR flake that's the only thing preventing merging, you can ping me or sherm1 to force it in anyway, you don't need to re-trigger.

BetsyMcPhail, 2025-01-14:
1/14 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1194

williamjallen, 2025-01-22:
1/22 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1232/consoleFull

jwnimmer-tri, 2025-01-29:
1/28:
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-experimental-leak-sanitizer/7054/consoleFull
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-debug/11393/consoleFull

BetsyMcPhail, 2025-02-04:
2/4:
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1300/
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-debug/977

BetsyMcPhail, 2025-02-12:
2/12 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1319

BetsyMcPhail, 2025-02-14:
2/14 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-debug/1010/

tyler-yankee, 2025-03-03:
3/3: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-debug/1041/

tyler-yankee, 2025-03-03:
3/3: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1360/

tyler-yankee, 2025-03-04:
3/4: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-release/1464

xuchenhan-tri, 2025-03-04:
3/4: [https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1373](https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1373)

xuchenhan-tri, 2025-03-05:
3/5: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1379

xuchenhan-tri, 2025-03-06:
3/6: https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-gcc-bazel-continuous-debug/2555/

BetsyMcPhail, 2025-03-17:
3/17 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1424

BetsyMcPhail, 2025-03-20:
3/20 https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-everything-release/1113

Aiden2244, 2025-03-27:
3/27 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1457/

tyler-yankee, 2025-04-07:
4/7: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1494/

BetsyMcPhail, 2025-04-08:
4/8 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1501/

BetsyMcPhail, 2025-04-09:
4/9 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-debug/2672/

SeanCurtis-TRI, 2025-04-15:
4/15 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-debug/2689/

BetsyMcPhail, 2025-04-29:
4/29 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-everything-release/1227/

BetsyMcPhail, 2025-05-05:
5/5
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/2744/
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/2745
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1573

jwnimmer-tri, 2025-05-06:
Maybe we should try adding `--experimental_remote_cache_lease_extension=yes` to see it that helps?

We momentarily had it for other reasons (https://github.com/RobotLocomotion/drake-ci/pull/324 -- remote download minimal) but ripped it out again when it didn't fix that problem.  Maybe it fixes this problem?

tyler-yankee, 2025-05-06:
Sounds like it's at least worth a shot. If that doesn't work, there might be also `--noexperimental_inmemory_dotd_files` and/or `--noexperimental_inmemory_jdeps_files`, though this might slow the builds down as it would write the `.d` / `.jdeps` files to disk instead of keeping them in memory (https://github.com/bazelbuild/bazel/issues/22854).

ggould-tri, 2025-05-07:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-gcc-bazel-continuous-debug/2755/

tyler-yankee, 2025-05-08:
I just doubled the size of the remote cache disk (1 TB -> 2 TB). FWIW, as I'm writing it's using 713 GB. 

Since this issue seems to occur in bunches (i.e. when multiple builds run at once), perhaps they're all trying to add files to the remote cache and "older" files that are actually still by other builds are getting evicted.

tyler-yankee, 2025-05-08:
I suppose another thing to look into could be ([docs](https://bazel.build/remote/caching)):

> Starting with Bazel 7.4, you can use `--experimental_disk_cache_gc_max_size` and `--experimental_disk_cache_gc_max_age` to set a maximum size for the disk cache or for the age of individual cache entries. Bazel will automatically garbage collect the disk cache while idling between builds; the idle timer can be set with `--experimental_disk_cache_gc_idle_delay` (defaulting to 5 minutes).

ggould-tri, 2025-05-08:
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-debug/12690/consoleFull

jwnimmer-tri, 2025-05-08:
> I suppose another thing to look into could be ... experimental_disk_cache_gc_max_size ...

That's for the local disk cache, not the remote cache.

tyler-yankee, 2025-05-12:
We did a bunch in the above PR, plus doubled the size of the disk for the cache. We haven't seen the error come up in a few days, so if we see it again, we'll turn on the new verbose logging option and start debugging.

Aiden2244, 2025-12-02:
12/1 https://drake-jenkins.csail.mit.edu/job/linux-noble-gcc-bazel-continuous-debug/617/

tyler-yankee, 2026-01-06:
We immediately had a continuous failure with `Missing digest`: https://drake-jenkins.csail.mit.edu/job/linux-noble-clang-bazel-continuous-everything-address-sanitizer/708/

But this is possibly/probably due to leftover blobs in the cache that hadn't had the new flag applied yet? I'd like to see a failure e.g. tomorrow before we resolve that the above PR doesn't fix.