RobotLocomotion/drake #21290: Add `GeometryId` to meshcat path API to `MeshcatVisualizer`

nepfaff, 2024-04-10:
**Is your feature request related to a problem? Please describe.**
This would be useful for changing the meshcat properties of a given Drake geometry. See [this](https://stackoverflow.com/questions/78293578/how-to-edit-alpha-transparency-for-new-gltf-models-programatically) StackOverflow post for context.

**Describe the solution you'd like**
`MeshcatVisualizer`should have an API (e.g. `get_meshcat_path`) that takes a `GeometryID` and returns the corresponding meshcat path name.

nepfaff, 2024-04-10:
@SeanCurtis-TRI

jwnimmer-tri, 2024-11-02:
Specifically, we the logic in `MeshcatVisualizer<T>::SetObjects` that converts a `GeometryId` and its `FrameId` into a `path` to be factored out and callable as a public function, something like `std::string MeshcatVisualizer<T>::GetMeshcatPath(GeometryId) const;`.