RobotLocomotion/drake #21320: Joints need an API that more directly support quaternions

SeanCurtis-TRI, 2024-04-17:
**Is your feature request related to a problem? Please describe.**
Related to issue #21215 and PR #21308.

Essentially, we have usages that care about distinguishing between joints that have quaternions in their state space and those that don't. As of the time this issue submission, there is really only one joint that has a quaternion: `QuaternionFloatingJoint`, but we expect to add `BallQuaternionJoint` (or some such thing). As of #21308, the code in `linear_distance_and_interpolation_and_provider.cc` is doing some runtime type checking to determine if the joint has a quaternion. This is brittle. Whoever adds a new joint with a quaternion would have to know about that runtime type checking and accommodate this particular use case, otherwise these joints will not be handled correctly.

**Describe the solution you'd like**
The proper solution is to define an API on the `Joint` class that derived classes have to directly handle and then for Drake code (like the example indicated above) to use the virtual methods instead of doing its own type checking. It has the benefit of automatically accommodating new classes that implement the interface and authors of joints don't have to hunting for possible uses of the joint outside of its virtual interface.

**Describe alternatives you've considered**
More of the same -- create a shopping list of places to look. Not a good option.

**Additional context**
Add any other context or screenshots about the feature request here.

jwnimmer-tri, 2024-04-17:
> Describe alternatives you've considered

FYI There is another (superior) alternative, described here as "Option 3": https://github.com/RobotLocomotion/drake/pull/21308#issuecomment-2057913971.

SeanCurtis-TRI, 2024-04-17:
Thanks; I'm promoting that to the main issue description.