RobotLocomotion/drake #21323: Convex (and Mesh) support for QueryObject::ComputeSignedDistanceToPoint

SeanCurtis-TRI, 2024-04-17:
[`QueryObject::ComputeSignedDistanceToPoint()`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#aa4c7754ca66a8593b182ae990643875c) has a sparsely populated support matrix. With [recent changes](https://github.com/RobotLocomotion/drake/pull/21147) in support for the convex hull for `Mesh` and `Convex` shapes, it would be good if we could fill the support matrix to include `Convex`, explicitly, and `Mesh` (implicitly as its convex hull).

This function has historically been unique as we needed to make sure that it produced good gradients *and* could successfully be evaluated w.r.t. `AutoDiffXd`. As such, the shapes that *are* supported resulted from a bunch of bespoke Drake code. The gaps are simply those specific point-queries that Drake hasn't implemented yet. We should do so for a convex hull.

**Describe the solution you'd like**
We need to implement a specific point-convex-hull query.

We are guaranteed a convex hull for `Mesh` and `Convex` types. But we need to perform signed distance to a point. This essentially means we need to implement our own version of GJK-EPA tailored to this particular problem (which makes it much simpler). We'll need:

1. Convert the convex hull representation (`PolygonSurfaceMesh`) into a data structure that will allow us to walk along the convex hull i a particular direction.
2. Given the query point, walk along the boundary of the polytope to find the closest point (inside or outside).
3. Do so in such a way that the gradient is well defined even with `T = AutoDiffXd` (i.e., no sqrt of displacement vector).

**Describe alternatives you've considered**
None; either we support it or we don't.

jwnimmer-tri, 2024-04-18:
When choosing the class design and algorithms here, planning to support both `double` and `AutoDiffXd` makes sense.

However, in terms of incremental implementation milestones, landing support for `double` first and `AutoDiffXd` in a second pass would benefit our most immediate downstream needs (where we only need `double` in the near-term).

DamrongGuoy, 2024-04-18:
Is this a relevant issue?

- #3220 

At the end of that issue, @xuchenhan-tri and I made these two points:

1. For triangle meshes, we can compute **unsigned** distance using [geometry/proximity/calc_distance_to_surface_mesh.h](https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity/calc_distance_to_surface_mesh.h)
```C+++
double CalcDistanceToSurfaceMesh(
    const Vector3<double>& p_WQ, const TriangleSurfaceMesh<double>& mesh_W,
    const Bvh<Obb, TriangleSurfaceMesh<double>>& bvh_W);
```
2. If the triangle meshes are water-tight, I believe we can assign the correct **sign** using the angle-weighted pseudonormals in this paper, but I have not tried it.
[J.A. Baerentzen](https://ieeexplore.ieee.org/author/37332808600); [H. Aanaes](https://ieeexplore.ieee.org/author/38527327700). [Signed distance computation using the angle weighted pseudonormal](https://ieeexplore.ieee.org/abstract/document/1407857). [IEEE Transactions on Visualization and Computer Graphics](https://ieeexplore.ieee.org/xpl/RecentIssue.jsp?punumber=2945) ( Volume: 11, [Issue: 3](https://ieeexplore.ieee.org/xpl/tocresult.jsp?isnumber=30527&punumber=2945), May-June 2005)

Additionally, if the mesh is tetrahedral, we can extract its water-tight triangle surface mesh ([ConvertVolumeToSurfaceMesh()](https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity/volume_to_surface_mesh.h#L65)) and then use the same code above.

I am more concerned about the return [grad_W](https://github.com/RobotLocomotion/drake/blob/master/geometry/query_results/signed_distance_to_point.h#L69) in the `struct` [SignedDistanceToPoint](https://github.com/RobotLocomotion/drake/blob/master/geometry/query_results/signed_distance_to_point.h#L24). I hope we can use the pseudonormal as a gradient. In general, there are multiple nearest points p_GN and multiple gradients grad_W when the query point is on the medial axis.

jwnimmer-tri, 2024-04-18:
> Is this a relevant issue?

Somewhat.  It *is* useful to cross-mention #3220 since it's related (so thank you for that), but that issue is about signed distance to non-convex meshes and this issue is about signed distance to (surely watertight) convex hulls.

jwnimmer-tri, 2024-05-03:
To make sure we're on the same page, possibly a good first step here would be to open a small Draft PR that contains the documentation changes for this new feature, and then after checking that over maybe we add a quick pseudo-code (as comments) for the new test case(s) that will demonstrate it working.

chen-tianjian, 2024-05-30:
Just a note from user perspective: we are interested in using the `ComputeSignedDistanceToPoint` function against **deformable** meshes. I would assume this effort for generic meshes should make that happen, but let's make a note to keep these ideas connected.

chen-tianjian, 2024-07-17:
We encountered this issue again in some other scenarios (not deformables). We do need to query point to mesh/convex distance, with regular rigid objects with mesh collision bodies. May I ask if there is a timeline estimate for this issue to be resolved? Thanks

DamrongGuoy, 2024-07-17:
I'm back from vacation and continue to work on it.  I'll try to have a prototype of `ComputeSignedDistanceToPoint` supporting mesh/convex in a week or two.  Then, cleaning up and PRs might take a few more weeks.

I'll keep you updated.  Thanks for asking.

DamrongGuoy, 2024-08-29:
I update that we have:

1. In-flight PR #21566 for the low-level calculation
2. My prototype [branch](https://github.com/DamrongGuoy/drake/tree/draft0_ProxEng_mesh_signed_distance) to use it in ProximityEngine. I'm making it into a PR.
3. User-facing documentation PR #21471

PR #21471 (2024-11-19): [geometry] ComputeSignedDistanceToPoint from meshes
Fix #21323.

We need two tasks:

- [x] #21566 
- [X] ComputeSignedDistanceToPoint from meshes in ProximityEngine (draft [branch](https://github.com/DamrongGuoy/drake/tree/draft0_ProxEng_mesh_signed_distance))

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21471)
closed 2024-11-19 by DamrongGuoy