RobotLocomotion/drake #21326: RenderEngineGlTest.MultiMaterialObj unit test failure vs nvidia-535 driver

jwnimmer-tri, 2024-04-18:
This test is failing locally on TRI workstations after the upgrade from `nvidia-driver-525` to `nvidia-driver-535`.

```
==================== Test output for //geometry/render_gl:internal_render_engine_gl_test:
Using drake_cc_googletest_main.cc

[==========] Running 32 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 32 tests from RenderEngineGlTest
...
[ RUN      ] RenderEngineGlTest.MultiMaterialObj
geometry/render_gl/test/internal_render_engine_gl_test.cc:477: Failure
Value of: CompareColor(expected_color_, color, inlier)
  Actual: false (At pixel (320, 240)
  Expected: (4, 240, 32, 255)
  Found: (111, 0, 204, 255)
  with tolerance: 1)
Expected: true
Color at: (320, 240)
Google Test trace:
geometry/render_gl/test/internal_render_engine_gl_test.cc:1205: multi-material test on green face - clone
...
```

Hopefully this is caused by an inconsistency in our implementation or test which we can work around, rather that some more fundamental bug in the driver itself.

jwnimmer-tri, 2024-04-25:
FYI The work-around to run all tests but skip the failing one is:
```
bazel test -- //... -//geometry/render_gl:internal_render_engine_gl_test
```

SeanCurtis-TRI, 2024-05-21:
For future reference, on useful technique for debugging these kinds of problems is to enable debugging in the `OpenGlContext`. To do so, simply edit https://github.com/RobotLocomotion/drake/blob/master/geometry/render_gl/internal_render_engine_gl.cc#L702 to pass `true` to the constructor of the `OpenGlContext`; this will enable debugging.

In this particular case, it ended up spewing a bunch of errors about a `RenderEngineGl` instance not properly owning the render targets that it was trying to manipulate. In the future, it may shed equally helpful light onto other problems.

PR #21466 (2024-05-21): Enforce uniqueness of render buffers between RenderEngineGl instances
Multiple RenderEngineGl instances share OpenGl objects in their shared context. Historically, this has included render buffers. A recent upgrade to NVidia drivers exposed this as problematic. Now, each clone is responsible for creating its own render targets.

Resolves #21326

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21466)
closed 2024-05-21 by jwnimmer-tri