RobotLocomotion/drake #21343: IrisInConfigurationSpaceFromCliqueCover can call IrisInConfigurationSpace with infeasible sample points

RussTedrake, 2024-04-21:
### What happened?

Reported by @AlexandreAmice on slack; I'm capturing it here to track.

Currently `IrisInConfigurationSpace` throws an error if the seed point is within `configuration_space_margin` of the obstacle. This can cause an issue in `IrisInConfigurationSpaceFromCliqueCover` since when asking for high coverage, we often find cliques that are fairly close to the configuration space obstacle.  @Michaelszeng is trying to rectify this issue but we have somewhat of a dilemma on this. I see two avenue forward and would love feedback/other ideas.

1. Sample only clique points at least `configuration_space_margin` away from obstacles. This would be achieved by adding `configuration_space_margin` padding to all collision pairs in the `collision_checker`. *Advantages:* very easy for the user. *Disadvantages:* This requires making the `collision_check` mutable in `IrisInConfigurationSpaceFromCliqueCover`  as we would need to change the padding in the collision checker. It also would requires (from the style guide) changing the argument order. It also would potentially slow down the function call (though its already a slow function so maybe not a big deal).
2. Discard cliques which hit this issue and log a warning to the user, potentially telling them to change the padding in their checker to be in line with `configuration_space_margin`. *Advantages:* we already ignore cliques which fail to compute a circumscribed ellipse (which can happen if the clique is in an affine subspace) plus we get to keep collision_checker a const reference. *Disadvantages:* a bit more involved for users and perhaps a bit more error prone.

cc @wernerpe

wernerpe, 2024-04-21:
Comment on solution 1. If we go that route we need to ensure that the center of the ellipsoid is checked with the same padding otherwise we wont solve the problem. (Even if all of the samples obey the margin, the center of the ellipse can be close to cspace holes we missed which would trigger the error.)

RussTedrake, 2024-04-23:
@wernerpe says that resolving #18830 could resolve this (because Iris could use the same collision checker padding used in the clique cover).

lunatic006, 2025-08-15:
Hi, I'm experiencing this issue too right now.

I want to add that even when we use the newer algorithms IrisNP2, IrisZO (which are supposed to use the coliisonchecker params to solve the issue) they still throw an error

I think its because when adding the polytope tangent to the closest collision, in `AddTangentToPolytope`, there's an extra sanity check that relies on the config_space_margin in iris_options for some reason, which throws an error. I think switching it to the correct margin in the collision checker's params is easy and would definitely result in the issue happening less often.

But, even when I set the two margins equal to each other I think it sometimes still throws the same error within 'AddTangentToPolytope' from the sanity check.