RobotLocomotion/drake #21351: Parallelize Intersection Checks in GcsTrajectoryOptimization

wrangelvid, 2024-04-23:
When adding a regions through GcsTrajectoryOptimization::AddRegions or edges through GcsTrajectoryOptimization::AddEdges, a lot of pairwise intersection checks are performed in series. These intersections checks can be moved into one single optimization problem which is then parallelized by the optimizer. However, since the checks habe been moved to [CalcPairwiseIntersections](https://github.com/RobotLocomotion/drake/blob/ee96c093c0b131c00d527cd053d74bc8465a1c14/geometry/optimization/geodesic_convexity.cc#L231), this may require more care.

AlexandreAmice, 2024-08-09:
Related issue is #19119. Perhaps resolving there would be more appropriate than having ad-hoc parallelization of solving MathematicalPrograms scattered throughout Drake. Though for intersection checks we might be able to do something better than the general case.

cohnt, 2024-10-30:
Dropping by to comment that I'm planning to get to this soon. I'd like to add parallelization in two places. 

1. The computation of AABBs of all the sets (since this preprocessing strategy already leads to a massive speedup, even without parallelization).
2. The actual set intersection programs.

The first needs a parallel for loop, but should be pretty straightforward. The latter can be done pretty easily with SolveInParallel.

PR #22142 (2024-11-12): Parallelize the Pairwise Set Intersection Checking for GCS (with or without Continuous Revolute Joints)
Closes #21351. (Although once we have a `parallelism` option added to GCS, as is being discussed in #21965, we should make sure that gets plumbed through.)

Note that the parallelism here is used in two ways -- the computation of AABBs of the constituent sets is handled with a parallel for loop, and then the pairwise intersections themselves are checked with `SolveInParallel`.

This leads to significant speedups, especially on faster graphs. With 224 sets (resulting in 5928 edges), we have t [... cut]