RobotLocomotion/drake #21352: Segmentation Fault in Pydrake

axbycc-mark, 2024-04-23:
### What happened?

The following script results in a Segmentation Fault. I'm surely doing something wrong, but I expect at least an error message instead of a segfault.

```
import numpy as np
import pydrake
from pydrake.multibody.parsing import Parser
from pydrake.multibody.plant import AddMultibodyPlantSceneGraph
from pydrake.systems.analysis import Simulator
from pydrake.systems.framework import DiagramBuilder
from pydrake.math import RigidTransform, RollPitchYaw
from pydrake.geometry import StartMeshcat
from pydrake.visualization import AddDefaultVisualization, ModelVisualizer

def create_scene(sim_time_step, meshcat):
    builder = DiagramBuilder()

    # construct a scene graph, and a multibodyplant
    # and wire their ports together the way Drake likes
    plant, scene_graph = AddMultibodyPlantSceneGraph(
        builder, time_step=sim_time_step)

    parser = Parser(plant)
    parser.AddModels(
        url="package://drake_models/ycb/003_cracker_box.sdf")
    parser.AddModels(
        url="package://drake_models/ycb/meshes/004_sugar_box_textured.obj")
    plant.Finalize()

    tx_world_sugar = RigidTransform(
        RollPitchYaw(np.asarray([0.0, 0.0, 0.0])), p=[0,0,0.5])
    tx_world_cracker_box = RigidTransform(
        RollPitchYaw(np.asarray([ 0.1, 0.2, 0.3 ])), p=[0,0,0.8])

    sugar_box = plant.GetBodyByName("004_sugar_box_textured")
    plant.SetDefaultFreeBodyPose(sugar_box, tx_world_sugar)

    cracker_box = plant.GetBodyByName("base_link_cracker")
    plant.SetDefaultFreeBodyPose(cracker_box, tx_world_cracker_box)

    AddDefaultVisualization(builder=builder, meshcat=meshcat)

    diagram = builder.Build()
    return diagram

def main():
    meshcat = StartMeshcat()

    print("Creating scene")
    diagram = create_scene(sim_time_step=0.01, meshcat=meshcat)

    print("Creating simulator")
    simulator = Simulator(diagram)
    simulator.Initialize()
    simulator.set_target_realtime_rate(1.0)

    print("Simulating")
    simulator.AdvanceTo(1.0)
    meshcat.PublishRecording()

    
    
if __name__ == "__main__":
    main()

```

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
INFO:drake:Meshcat listening for connections at http://localhost:7000
Creating scene
Creating simulator
Simulating
Segmentation fault (core dumped)
```

jwnimmer-tri, 2024-04-24:
Thanks, I can confirm I can reproduce the problem.  The fault happens during `PublishRecording`, somehow.

jwnimmer-tri, 2024-04-24:
The problem is related to missing calls to Start/Stop recording.  This variation runs OK for me:

```py
...
    print("Creating simulator")
    simulator = Simulator(diagram)

    print("Simulating")
    meshcat.StartRecording()
    simulator.AdvanceTo(1.0)
    meshcat.StopRecording()
    print("Publishing")
    meshcat.PublishRecording()
    print("Done")
```

I agree it shouldn't segafult, though!  We'll investigate and try to provide a better error message.

PR #21360 (2024-04-26): [geometry] Meshcat has a default recording
Having an occasionally-null recording is a segfault beacon. Either we need to add more nullptr checks and nice error messages, or just do the obvious thing and have a non-null default.

Closes #21352, and has some chance of making #21279 simpler.
 
+@DamrongGuoy for feature review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21360)
closed 2024-04-26 by SeanCurtis-TRI