RobotLocomotion/drake #21420: //geometry/render_gl:multithread_safety_test failure on Ubuntu 24.04

jwnimmer-tri, 2024-05-09:
### What happened?

In both CI and locally, our two unit tests that use multithreaded OpenGL are failing.

Here's the `//geometry/render_gl:multithread_safety_test` output (below).  The `//systems/sensors:rgbd_sensor_async_gl_test` output is equivalent.

### Version

2b46e26f (present day)

### What operating system are you using?

Ubuntu 24.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
[ RUN      ] ThreadSafetyTest.TextureLibrary
MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
glx: failed to create drisw screen
[       OK ] ThreadSafetyTest.TextureLibrary (127 ms)
[ RUN      ] ThreadSafetyTest.OpenGlContext
X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  150 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  127
  Current serial number in output stream:  127
```

jwnimmer-tri, 2024-05-13:
I did some more debugging.  A few observations:
- When X experiences an error (e.g., during `glXMakeCurrent`) it calls `exit()`.  The test was anticipating an exception, not death.  I'm not sure where/what changed compared to prior versions of the libraries.
- Even with that fixed (i.e., calling `XSetErrorHandler` pointing to our own non-exiting handler), we still have problems.  In `RenderEngineGl::~RenderEngineGl()` the call to `opengl_context_->MakeCurrent()` can throw, and throwing inside a destructor is a `std::terminate`.

Reading the [docs for glXMakeCurrent](https://registry.khronos.org/OpenGL-Refpages/gl2.1/xhtml/glXMakeCurrent.xml) its very clear that if the context is current on another thread already, trying to make it current on a new thread (without clearing it on the other thread) shall be an error.  As best I can tell, we have _nothing_ in our code that attempts to clear current in a way that meets the spec.  This seems like a bug that we've had forever, but now finally 24.04 is calling us on.

---

BTW Yet again, `gdb` is again _much improved_ on 24.04 compared to 22.04.  Now it offers to download debug symbols from Ubuntu, so I was able to get function names and line numbers in the backtrace.

SeanCurtis-TRI, 2025-10-17:
Nope. CI is *not* as happy as my puget. So, we have apparently lost the ability to reproduce this locally, but we're stuck in the world where CI is still cranky about it.

I may have to play with downgrading my nvidia driver.

jwnimmer-tri, 2025-10-17:
It's possible (likely?) that when I reported being able to repro locally, it was on my laptop with intel graphics (not nvidia).

SeanCurtis-TRI, 2025-10-17:
Yep. I just confirmed failure on my laptop. (Huzzah!)