RobotLocomotion/drake #21462: Noble Provisioned AMI

jwnimmer-tri, 2024-05-20:
Towards #21335.

Please create an AMI from the current `install_prereqs`.  We'll use it for some Noble Provisioned jobs coming up.

(FYI even though some tests are currently failing in Noble Unprovisioned, they do not indicate any problem with `install_prereqs`; we'll be fixing them through other means.)

jwnimmer-tri, 2024-05-20:
(Note that I'll also soon be posting the list of Noble Production builds to spin up, so you could wait to do this until just before we need it. Or do this one first, fine by me either way.)

BetsyMcPhail, 2024-05-21:
The noble provisioned AMI has been created and add to Jenkins. It is ready to use for provisioned jobs when they are created.

closed 2024-05-21 by BetsyMcPhail