RobotLocomotion/drake #21472: Mimic joints in SDF throws warning but also crashes parser

lcbw, 2024-05-22:
### What happened?

Labels:  component: parsing  configuration: linux

On attempt to parse the robotiq 85 gripper URDF as can be found in the apt package ros-humble-robotiq-description into Drake, warnings were thrown against using mimic joints with Drake, but then the file failed to parse anyway. Commenting out the mimic joints resolved the problem, indicating that the mimic joints aren't being ignored like the warning said they would be if they had been left in. [or something similar]

Example warning: 
Joint 'robotiq_85_right_knuckle_joint' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.

Example file which failed to parse:

```
<?xml version="1.0" ?>
<sdf version="1.7">
  <world name="gripper_test">
    <include>
      <uri>package://mast/descriptions/robotiq_testing/ur10e.urdf</uri>
      <name>robot</name>
    </include>

    <joint name="world_to_robot" type="fixed">
      <parent>world</parent>
      <child>robot::base_link</child>
    </joint>
    <include>
      <uri>package://mast/descriptions/robotiq_testing/gripper.urdf</uri>
      <name>gripper_body</name>
      <pose relative_to="robot::tool0">0 0 0 0 0 0</pose>
    </include>
    <joint name="robot_to_gripper_body" type="fixed">
      <parent>robot::tool0</parent>
      <child>gripper_body::ur_to_robotiq_link</child>
      <pose relative_to="robot::tool0">0 0 0 0 0 0</pose>
    </joint>
  </world>
</sdf>
```  

### Version

1.27.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
WARNING:drake:/home/user/workspaces/my_ws/install/my_ws/share/my_ws/descriptions/robotiq_testing/gripper.urdf:128: warning: Joint 'robotiq_85_right_knuckle_joint' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.
WARNING:drake:/home/user/workspaces/my_ws/install/my_ws/share/my_ws/descriptions/robotiq_testing/gripper.urdf:145: warning: Joint 'robotiq_85_left_inner_knuckle_joint' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.
WARNING:drake:/home/user/workspaces/my_ws/install/my_ws/share/my_ws/descriptions/robotiq_testing/gripper.urdf:152: warning: Joint 'robotiq_85_right_inner_knuckle_joint' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.
WARNING:drake:/home/user/workspaces/my_ws/install/my_ws/share/my_ws/descriptions/robotiq_testing/gripper.urdf:159: warning: Joint 'robotiq_85_left_finger_tip_joint' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.
WARNING:drake:/home/user/workspaces/my_ws/install/my_ws/share/my_ws/descriptions/robotiq_testing/gripper.urdf:166: warning: Joint 'robotiq_85_right_finger_tip_joint' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.
Traceback (most recent call last):
  File "/snap/pycharm-community/378/plugins/python-ce/helpers/pydev/pydevd.py", line 1534, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/snap/pycharm-community/378/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/user/workspaces/my_ws/src/my_ws/scripts/gripper_test.py", line 313, in <module>
    main()
  File "/home/user/workspaces/my_ws/src/my_ws/scripts/gripper_test.py", line 245, in main
    rdb.parser().AddModels(FindResourceOrThrow(args.resource))
RuntimeError: error: Child frame with name[gripper_body::ur_to_robotiq_link] specified by joint with name[robot_to_gripper_body] not found in world with name[gripper_test].
```

jwnimmer-tri, 2024-05-22:
Thanks for your report!  I see that the ROS apt package `ros-humble-robotiq-description` contains the Xacro input files, but not the actual URDF file literally.  In order for us to reproduce the problem (without having to guess how you ran Xacro), could you please attach the exact URDF file that you were trying to parse?  Thanks.

lcbw, 2024-05-23:
[gripper.txt](https://github.com/RobotLocomotion/drake/files/15418431/gripper.txt)
you'll need to rename this to gripper.urdf but interestingly enough github does not allow you to attach .urdf files

closed 2026-11-16 by rpoyner-tri