RobotLocomotion/drake #21475: [ci] Config changes for Noble support

jwnimmer-tri, 2024-05-23:
Changes to pre-merge:
- [ ] Replace `linux-jammy-clang-bazel-experimental-release` with the noble flavor instead
- [ ] Replace `linux-jammy-gcc-bazel-experimental-release` with the noble flavor instead

Changes to Continuous Production:
- [ ] Replace `linux-jammy-clang-bazel-continuous-release` with the noble flavor instead (concurrently with pre-merge change, to keep them aligned)
- [ ] Replace `linux-jammy-gcc-bazel-continuous-release` with the noble flavor instead (concurrently with pre-merge change, to keep them aligned)

Changes to Nightly Production:
- TBD

Changes to Weekly Production:
- TBD

jwnimmer-tri, 2024-05-23:
FYI @BetsyMcPhail I just cross-checked the above; no changes.

BetsyMcPhail, 2024-05-24:
Noble jobs have not been automatically added to https://github.com/RobotLocomotion/drake/tree/jenkins-jobs-experimental, need to investigate before closing this issue

jwnimmer-tri, 2024-05-24:
Interesting.  Right now, it duplicates the Continuous job, doesn't it?  I'm not sure what I was thinking.  So yes, let's use Everything for the nightly one.

closed 2024-05-28 by jwnimmer-tri