RobotLocomotion/drake #21484: Need to document the MultibodyPlant generalized_contact_forces output port

RussTedrake, 2024-05-25:
### What happened?

Follow-up from [this SO post](https://stackoverflow.com/questions/78525271/how-to-use-generalized-contact-forces-in-drake-multibodyplant), I was also unable to find any documentation describing the values available on this output port.

jwnimmer-tri, 2024-06-22:
One related question -- is there any reason we have a per-model output port for this, but not a whole-plant output port?  Maybe that nobody asked for one?

It's an outlier as the only per-model output that doesn't have a whole-plant neighbor.  In the current Calc function, we are already computing the whole-plant vector and then throwing way the part we didn't need.

staycoolish, 2025-04-28:
Are there any C++ tutorials or code samples that we can utilize for such functions? I want to read/publish contact forces, but I couldn't find any example in drake/examples folder (I am working with Kuka for ease of adaptation).