RobotLocomotion/drake #21514: Unstable plane-on-plane contact when switching over to contact models Lagged / Similar

jbinagia, 2024-06-03:
### What happened?

As discussed in this [slack thread](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1716474655591939) and reproduced in this [deepnote notebook](https://deepnote.com/workspace/drake-243a-4ff108f2-1127-4473-b0a6-96edd49a6e86/project/Tutorials-Duplicate-82c0dd10-08a8-4538-9b52-fdac87b63218/notebook/unstable_plane_on_plane-4e6ddf1007ca4a60903c9cae5c61b277), I discovered an issue whereby unstable plane-on-plane contact was exhibit for two floating models stacked on top of one another when using the Lagged or Similar contact models. A video illustrating this behavior is shown below: 

https://github.com/RobotLocomotion/drake/assets/30089571/81d11981-299e-49ab-8e9d-445cfb47524c

The hydroelastic moduli for the boxes are ~1e6 Pa. Because I do not see this instability for SAP, I expected Lagged and Similar to also be stable for this choice of moduli (i.e. I expect the same modulus value to be stable regardless of contact model choice). I believe this may be a result of the fact that SAP regularizes contact in the near-rigid regime, while Lagged and Similar models do not.

amcastro-tri, 2024-06-04:
This is an animiation of the instability showing the contact patch between the boxes.
This is the same typical problem we get with (multi) point contact at (large) discrete time steps.
With `kSap` the problem is greatly minimized, however it does not go away, you just need to run for longer becase it's a slower frequiency (~1Hz for kSap and ~15Hz for kSimilar). 
The reason is that `kSap` overestimates the amount of regularization at large dissipation values (in this case 0.1 sec). 
This, as explained in the paper above, brings other problems we'd like to avoid. So regularization is only hiding a common problem, it is not the solution. I'll post later a solution that seems to work very well.

[image]
**Instability problem for `kSimilar` with `E=1e6 Pa`, `d=20 s/m` and `dt=0.01 s`.**

amcastro-tri, 2024-06-05:
All right, teaser progress on this. 
The sim below is run with `kSimilar`. Just to make it a "bit" more difficult, I am dropping the box from a far height, the hydro modulus is `1.0e8 Pa` (there's nor regularization nor softening here, this is the real deal), no dissipation (to make it yet more difficult, so that any plaussible artifacts do not get damped out), the coefficient of friction is as before 0.4. Both boxes and the ground are compliant (with that very large modulus). As ou can see, there are no instabilitites. I run several initial conditions, orientatioins, etc and still I get this super nice stable sims.

I'll create a branch with the solution and link here shortly.

[image]

amcastro-tri, 2024-06-05:
Solution implemented in #21528.

jbinagia, 2024-06-05:
@amcastro-tri Just to confirm, this fix also works when using `kLagged` as well?

amcastro-tri, 2024-06-05:
yes, this improves time coherence regardless of the contact approximation, so it makes things better for kSap, kSimilar, kLagged and even kTamsi.

amcastro-tri, 2024-08-12:
#21773 introduces margin for hydroelastic contact to fix this issue.

Users must set a non-zero value of of the default in `DefaultProximityProperties::margin`. The recommended value is `0.1 mm` (`10⁻⁴ m`). 

After some experimentation our intentiion is to make this value of `10⁻⁴ m` the default value in Drake.

closed 2024-08-12 by amcastro-tri