RobotLocomotion/drake #21535: The `new_release` script reports the incorrect version for ros_xacro_internal

BetsyMcPhail, 2024-06-06:
### What happened?

While doing the monthly external upgrades as part of #21507, `bazel-bin/tools/workspace/new_release` reported that: `ros_xacro_internal needs upgrade from 2.0.10 to xacro-1.7.5` The version to upgrade to should be 2.0.11.

jwnimmer-tri, 2024-06-06:
Best guess, we need to update this pattern to ignore those really old tags that start with `xacro-...`:
https://github.com/RobotLocomotion/drake/blob/7228fd95e301bf0bea212bca23c60b42a6989b63/tools/workspace/new_release.py#L78

BetsyMcPhail, 2024-06-06:
Related to this, I've been caught twice by the Gymnasium update to `v1.0.0a1`. The new_release script should ignore alpha/beta/pre-releases.

mwoehlke-kitware, 2024-06-07:
> The new_release script should ignore alpha/beta/pre-releases.

We do that for `["alpha", "beta", "rc", "pre"]` in `_is_ignored_tag`. The parsing is fairly dumb (it just checks if those strings are *anywhere* in the commit), so I don't think you'd want to add `a` to the global ignore list, but this looks like a good case for adding a rule to `_IGNORED_TAGS`.

PR #21635 (2024-06-27): Add more tags to ignore to the new_release script
Resolves  #21535

Prior to these updates, the output from the `new_release` script was:
```
abseil_cpp_internal needs upgrade from ed34153e0d9cd15f9b9eb45e86b457e0a495aeea to 0ccc51f9ddbb407d579f8158d5421fbf3eea0524
crate_universe may need upgrade
gymnasium_py needs upgrade from v0.29.1 to v1.0.0a2
gz_math_internal needs upgrade from gz-math7_7.4.0 to gz-math7_7.5.0
msgpack_internal needs upgrade from cpp-6.1.1 to c-6.0.2
mypy_internal needs upgrade from v1.10.0 to v1.10.1
pycodestyle needs upgrade from 2.11.1 to 2.12.0
ros_xacro_internal needs [... cut]
closed 2024-06-27 by EricCousineau-TRI