RobotLocomotion/drake #21552: Getting `UpdateHessianType()` warnings in Drake 1.29.0

sea-bass, 2024-06-09:
### What happened?

When I upgraded from Drake 1.28.0 to 1.29.0, I started getting these really loud warnings printed to my console:

```
WARNING:drake:UpdateHessianType(): Unable to determine Hessian type of the Quadratic Constraint. Falling back to indefinite Hessian type.
```

Nothing else seems to have been affected, but I'd like to be able to resolve these in some way. Are there any resources for what this may mean, and how to fix it?

### Version

1.29.0

### What operating system are you using?

Ubuntu 24.04

### What installation option are you using?

pip install drake

sea-bass, 2024-06-09:
Update: I could get around it by being very explicit...

```python
                from pydrake.solvers import QuadraticConstraint
                prog.AddQuadraticConstraint(
                    0.5 * (x[k, n] + x[k + 1, n])
                    + (h[k] / 8.0) * (x_d[k, n] - x_d[k + 1, n]) 
                    - xc[k, n],
                    0.0, 0.0, QuadraticConstraint.HessianType.kIndefinite)
```

hongkai-dai, 2024-06-09:
@sea-bass as you have found out, explicitly specifying the Hessian type would resolve your problem. I filed the PR #21553

PR #21553 (2024-07-13): Improve the warning message when the Hessian type is indefinite.
Resolves #21552

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21553)
closed 2024-07-13 by hongkai-dai