RobotLocomotion/drake #21557: Upgrade to XCode 15

jwnimmer-tri, 2024-06-10:
**Is your feature request related to a problem? Please describe.**

In #21529, we discovered that the standard library in XCode 14 is missing some support for spaceship (`<=>`).

**Describe the solution you'd like**

Upgrade our macOS Ventura CI image(s) to **XCode 15**.

Of course then also we need to update our docs to reflect current CI, e.g.:
- https://drake.mit.edu/installation.html
- https://drake.mit.edu/from_source.html

**Describe alternatives you've considered**

Pollute our code with backcompat work-arounds.

**Additional context**

This is not an emergency.  It can wait until next week.

rpoyner-tri, 2024-06-12:
I don't know that Ventura and XCode 15 are compatble. I could be that the best Ventura could host is XCode 14.3.1. From the evidence I have so far, that version doesn't solve our spaceship problems.

jwnimmer-tri, 2024-06-12:
Does XCode 15.2 have a sufficient spaceship?

_Xcode 15.2 requires a Mac running macOS Ventura 13.5 or later._
-- https://developer.apple.com/documentation/xcode-release-notes/xcode-15_2-release-notes

rpoyner-tri, 2024-06-12:
Apparently XCode 15.2 does not have sufficient spaceship, based on my experiments so far. Details below.

* Installed XCode 15.2 from manually downloaded .xip file on an internal M2 machine running Ventura 13.6.7.
* Tried to build a reduced test case (see below); failed with errors like those seen before.

```
diff --git a/common/test/type_safe_index_test.cc b/common/test/type_safe_index_test.cc
index d546e9d908..11708d7ae5 100644
--- a/common/test/type_safe_index_test.cc
+++ b/common/test/type_safe_index_test.cc
@@ -1,6 +1,8 @@
 #include "drake/common/type_safe_index.h"
 
+#include <compare>
 #include <limits>
+#include <optional>
 #include <regex>
 #include <sstream>
 #include <string>
@@ -605,6 +607,18 @@ GTEST_TEST(TypeSafeIndex, SortedPairIndexHashable) {
   EXPECT_TRUE(pairs.contains(SortedPair<AIndex>(a1, a2)));
 }
 
+GTEST_TEST(TypeSafeIndex, OptionalCompare) {
+  struct Thing {
+    std::optional<AIndex> field;
+    auto operator<=>(const Thing&) const = default;
+  };
+  Thing thing1;
+  thing1.field = AIndex(1);
+  Thing thing2;
+  thing2.field = AIndex(2);
+  EXPECT_TRUE(thing1 < thing2);
+}
+
```

Among the frustrations (never mind the ridiculously bad installation processes), there is no convenient way to get a reliable version number out of Xcode. You'd hope that `clang --version` would be useful, but it is the same sometimes across different Xcode versions. Internet advice promised good things from `softwareupdate --history`, but it is completely oblivious to "side-loading" -- installing from downloaded archives instead of using GUI updaters (which often won't do what you want).

So, today's version extraction command recommendation is something like:
```
$ cat /Applications/Xcode.app/Contents/version.plist |grep -A1 ShortVersion |tail -n1 |sed 's#.*>\(.*\)<.*#\1#g' 
```

jwnimmer-tri, 2024-06-12:
I suppose if "XCode 15.x" is not enough of a precise victory condition, then the next question is which version of XCode _is_ sufficiently new enough?  You have a successful result with ... 15.4?  There's not that many versions newer than 15.2, so I guess it's a recent fix.  None of Apple's release notes seem to mention this.

rpoyner-tri, 2024-06-13:
15.3 seems to be good enough, but that is not documented to support being hosted on Ventura.

jwnimmer-tri, 2024-06-13:
Even if we could force 15.3 into CI, it seems like asking users to install it using an off-nominal procedure would be too risky of a proposition.  I think that means we max out at 15.2 until October 2024, when we drop support for Ventura.

It sounds like 15.2 doesn't actually help us, really?  In that case, we should close this issue and save the work of CI upgrades.

closed 2024-09-09 by jwnimmer-tri