RobotLocomotion/drake #21576: Ubuntu Noble APT repo missing?

elrom, 2024-06-17:
### What happened?

Following the instruction [here](https://drake.mit.edu/apt.html#stable-releases), results in apt complaining about a 403.

In a browser, this is confirmed.
https://drake-apt.csail.mit.edu/noble/dists/noble/InRelease results in a 403, whereas https://drake-apt.csail.mit.edu/jammy/dists/jammy/InRelease works as expected.

jwnimmer-tri, 2024-06-17:
Sorry, we just added 24.04 support 2 days ago and not everything has been uploaded yet.

For now, you can manually `dpkg -i` from the release attachment:
- https://github.com/RobotLocomotion/drake/releases/download/v1.30.0/drake-1.30.0-noble.tar.gz

elrom, 2024-06-17:
Oh wow, I didn't realize. Thanks for the super prompt reply!

closed 2024-06-18 by jwnimmer-tri