RobotLocomotion/drake #21577: Numpy 2.0 supports

jwnimmer-tri, 2024-06-17:
**Is your feature request related to a problem? Please describe.**

NumPy 2.0 has been released.  See https://numpy.org/devdocs/numpy_2_0_migration_guide.html.

**Describe the solution you'd like**

Drake should be compatible with both NumPy 1.x and 2.x.

jwnimmer-tri, 2024-06-17:
For starters we need to merge up our pybind11 fork to incorporate https://github.com/pybind/pybind11/pull/5050, i.e., to v2.12.0.

jwnimmer-tri, 2024-06-17:
To unbreak our wheel builds, I believe can do something alone these lines:

```patch
--- a/tools/wheel/image/provision-python.sh
+++ b/tools/wheel/image/provision-python.sh
@@ -42,7 +42,7 @@ ln -s /usr/local/bin/python /usr/bin/python
 # keeping them up to date if they are?
 pip install \
     matplotlib \
-    numpy \
+    'numpy < 2' \
     pyyaml \
     semantic-version \
     setuptools \
```

ggould-tri, 2024-06-17:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-unprovisioned-gcc-wheel-nightly-release/111 is this issue; I am using the magic word `//:install` in this comment so that future buildcop searches will find this issue when they search for the failing target.

jwnimmer-tri, 2024-06-18:
I think possibly the last remaining obstacle is to figure out how to rework https://github.com/RobotLocomotion/pybind11/commit/0da7ac7bda9d0cecd6a576e47afcfd4954f4bfcc to no longer rely on `itemset`, which is related to https://github.com/pybind/pybind11/issues/1181.

jwnimmer-tri, 2024-06-20:
The first thing to try is if we can call [`__setitem__`](https://docs.python.org/3/reference/datamodel.html#object.__setitem__) with a slice, instead of `itemset`.

bernhardpg, 2024-07-03:
FYI If I now try to do a fresh install of Drake and install all the prereqs I am still running into this issue.

I run:
```
python3 -m venv drake_env
source drake_env/bin/activate
cd drake
./setup/mac/install_prereqs.sh
```

While I don't get an error from the above setup, I do run into the ```AttributeError: `itemset` was removed from the ndarray class in NumPy 2.0.``` error when I try to build and install Drake from source.

It seems that brew is still installing NumPy 2.0 and that `numpy < 2` in `requirements.txt` only affects a different numpy install (installed by pip) which is not the one used when building.

I can run succesfully run `brew uninstall numpy` but then trying to build and install with Bazel gives the new error (it does not help to run `bazel clean --expunge`)

```
AttributeError: module 'numpy' has no attribute 'lib'
```

Is there a current workaround for this? Or is it currently not possible to build Drake from source on mac?

bernhardpg, 2024-07-04:
**Update/temporary fix:** I found that applying [this](https://github.com/RobotLocomotion/drake/pull/21677/files) change WITHOUT setting up in a virtual environment seems to have worked without any issues for me (so far)

jwnimmer-tri, 2024-07-09:
On the Drake master branch as of just now, we hope that numpy 2.0 works correctly.

Tomorrow's nightly wheel build (i.e., `drake >= 0.0.20240710`) should also work.

Next week's stable release 1.31.0 should also work.

---

The actions remaining on this ticket are to investigate and clean up a couple of CI tests that we had to disable in order to pass.

PR #22112 (2024-11-06): [bindings] Re-enable some gym test cases on macOS
This involves deferring the import of the non-hermetic dumpster fire known as `matplotlib.pyplot` until it is needed, so that unit tests are not exposed to it.

Closes #21577.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22112)