RobotLocomotion/drake #21588: Deepnote tutorials links are broken

jwnimmer-tri, 2024-06-18:
On the https://drake.mit.edu/ landing page, the 3 links to tutorials notebooks are broken.

In the deepnote project on the Index notebook, the links to all of the other tutorials are broken.

We previously used to be able to link to e.g. `authoring_leaf_systems.ipynb` and have it find the correct file.  That style of link appear to no longer work.

+@RussTedrake offered to email Deepnote about this.

RussTedrake, 2024-06-18:
I know that back in December, I had to change the way that I was automating the deepnote links in my coursenotes: https://github.com/RussTedrake/htmlbook/commit/95a49ae5a332bbea3bde45d167cc2fc06626ad61 

The notebooks are assigned a unique ID when they are uploaded, which can be used to generate the link.  Those links have been stable since that time.  I believe that a one-time sweep through the index notebook could update all of those broken links, and they would stay fixed (until the next time Deepnote changes something like this)...

jwnimmer-tri, 2024-06-18:
As of the recent 1.30.0 deployment, here is the link that worked for `dynamical_systems.ipynb`:

https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/notebook/dynamical_systems-f5cb4b4cfc364799bad24282d75628a3

I just did the same deployment step again (same docker image base) for that one single file.  Now the link is different:

https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/notebook/dynamical_systems-f04b96da5f714ef093bb9be47c8dbf8c

For the Drake tutorials deployment, we copy the new notebook with a `-2` suffix, then delete the prior copy, then rename the new file to take over the old name.  So the unique ID changes every month.

jwnimmer-tri, 2024-06-18:
Aha, but the old link still (for now?) redirects to

https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/notebook/f04b96da5f714ef093bb9be47c8dbf8c

which seems to work.  I wonder if that is reliable.

jwnimmer-tri, 2024-06-18:
> I believe that a one-time sweep through the index notebook could update all of those broken links, ...

We reset that notebook upon every release, so that any _edits_ to the index get deployed.  Manually patching in Deepnote-specific links in the index notebook would mean that we need to have some way to notice/merge edits to the index.

In any case we can look at whatever automation makes the most sense, but it would help to know the starting point.  Can Deepnote fix vanilla relative links to other `*.ipynb` files (e.g., `- [Authoring a Multibody Simulation](./authoring_multibody_simulation.ipynb)`) to please work again?  That would make Deepnote and pip-local back to being on par, which would greatly simplify our life.

RussTedrake, 2024-06-23:
https://community.deepnote.com/c/ask-anything/relative-links-from-markdown-between-notebooks