RobotLocomotion/drake #21624: Proximity reification during algebraic loop detection

SeanCurtis-TRI, 2024-06-25:
### What happened?

When `DiagramBuilder` queries for algebraic loops, this can lead to the allocating a default context on an `Expression`-valued `Diagram`. When `SceneGraph` allocates a default context, its default proximity properties are assessed and proximity geometries have the potential to get reified. This is wasted work.

The proximal cause is that the calculation of direct feedthrough for `SceneGraph` is not resolved by examining the dependency graph and instantiating `System<Expression>` and its default `Context<Expression>` is the fallback.

The solution would be to make sure that `SceneGraph`'s output ports explicitly declare their dependencies. Right now it defaults to the "all sources" ticket which is treated by the feed through calculation as meaningless. Simply declaring it dependent on parameter (its `GeometryState`) and all of its inputs would be sufficient.

jwnimmer-tri, 2024-06-29:
Fixed in #21630.

closed 2024-06-29 by jwnimmer-tri

Xining-Du, 2024-08-07:
Just curious which change(s) caused this issue?

We used to use Drake 1.27 and didn't see this problem. However it started to appear after we imported Drake 1.30, and we had to cherrypick #21630 as a workaround. So presumably it is related to changes between 1.27 and 1.30?