RobotLocomotion/drake #21673: FCL bug in IK optimization with AddMinimumDistanceLowerBoundConstraint

nepfaff, 2024-07-02:
### What happened?

The following error:
```
RuntimeError: Exception while evaluating SNOPT costs and constraints: 'Error with configuration
  Original error message: bazel-out/k8-opt/bin/external/fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
```
while solving an IK non-penetration projection problem with SNOPT.

I pasted the full error log for reproducability.

### Version

1.30.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
RuntimeError: Exception while evaluating SNOPT costs and constraints: 'Error with configuration
  Original error message: bazel-out/k8-opt/bin/external/fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.027183880000000000543, -0.021052830000000001542, -0.013271629999999999538),
      Vector3<double>(-0.029787089999999998707, -0.018163209999999999078, 0.0023578100000000001141),
      Vector3<double>(-0.029523460000000001369, -0.014922700000000000548, 0.0079630499999999992955),
      Vector3<double>(0.018158350000000000185, -0.028047450000000001546, -0.016771020000000001066),
      Vector3<double>(-0.033257000000000001838, -0.0034294899999999998823, -0.02402831000000000064),
      Vector3<double>(0.030779520000000001184, -0.016741050000000000375, -0.0062909699999999998954),
      Vector3<double>(0.031590300000000001712, -0.011201890000000000805, 0.0022611499999999999655),
      Vector3<double>(0.030848339999999998401, -0.016967369999999998981, -0.013022880000000000567),
      Vector3<double>(0.0089922000000000005399, -0.005754650000000000043, -0.039821399999999999852),
      Vector3<double>(0.0084664200000000005425, -0.0017271199999999999573, -0.04059970999999999719),
      Vector3<double>(-0.00074333999999999999467, -0.0034554999999999998044, -0.042391610000000003178),
      Vector3<double>(-0.0031593099999999998094, -0.0056862400000000003913, -0.042894870000000001609),
      Vector3<double>(0.028471980000000000899, -0.017014069999999999194, -0.019219239999999998547),
      Vector3<double>(0.019703479999999998912, -0.0068651700000000002999, -0.034910230000000000428),
      Vector3<double>(-0.0061926200000000002424, -0.0083001399999999992546, -0.04281131999999999993),
      Vector3<double>(0.014279659999999999576, -0.032117050000000001153, -0.010280209999999999695),
      Vector3<double>(-0.00242264000000000021, -0.028708859999999999241, 0.013088809999999999334),
      Vector3<double>(-0.0227650299999999986, -0.02248137000000000052, 0.0086427499999999993913),
      Vector3<double>(-0.0008474599999999999668, -0.0094668600000000005523, 0.059735150000000000747),
      Vector3<double>(0.0099082100000000006279, -0.032700279999999998237, -0.00028732999999999997475),
      Vector3<double>(0.0054166600000000002246, -0.032023179999999998313, 0.003964360000000000106),
      Vector3<double>(-0.015149090000000000472, -0.030018500000000000044, 0.0038727300000000000092),
      Vector3<double>(-0.013153349999999999417, -0.028098009999999999653, 0.0098451100000000006662),
      Vector3<double>(-0.0053969600000000001253, -0.031911280000000000201, 0.0046099399999999998712),
      Vector3<double>(-0.018971149999999999125, -0.026639590000000001185, 0.0068275799999999997603),
      Vector3<double>(0.0087256299999999998446, -0.033899480000000002666, -0.0082973800000000000332),
      Vector3<double>(0.013405669999999999739, -0.0027963400000000000235, 0.054792069999999998275),
      Vector3<double>(0.031803619999999997725, -0.0015181899999999999909, 0.0080110000000000007592),
      Vector3<double>(0.032492689999999997685, -0.0061319699999999996085, 0.0053189999999999999433),
      Vector3<double>(0.035098669999999998481, 0.0117753500000000004, -0.011222400000000000431),
      Vector3<double>(-0.031461019999999999264, 0.0024656800000000000418, -0.026877829999999998389),
      Vector3<double>(-0.037982410000000001171, -0.005085420000000000218, -0.0077946800000000000114),
      Vector3<double>(-0.035593800000000001826, -0.0088536900000000008648, 0.0020318599999999999162),
      Vector3<double>(0.033346519999999997552, -0.012396209999999999521, -0.009431099999999999553),
      Vector3<double>(0.034542349999999999444, -0.0068445700000000003413, -0.004166679999999999863),
      Vector3<double>(0.033087320000000003395, 0.014188879999999999201, -0.016763059999999999766),
      Vector3<double>(0.034359969999999996626, 0.0071343600000000001807, -0.018503900000000000209),
      Vector3<double>(0.036166349999999999887, 0.0066439500000000000293, -0.01100461999999999968),
[... 233 lines cut ...]
      Vector3<double>(0.0019087799999999998952, 0.031599679999999998048, -0.021888689999999998842),
      Vector3<double>(0.0094769299999999997292, 0.033574489999999998502, -0.014093150000000000469),}),
    542,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 65, 154, 6, 3, 65, 6, 28, 3, 23, 61, 18, 3, 154, 218, 6, 3, 26, 65, 28, 3, 229, 1, 128, 3, 17, 24, 18, 3, 228, 24, 17, 3, 1, 228, 17, 3, 229, 228, 1, 3, 28, 6, 34, 3, 158, 0, 7, 3, 12, 0, 3, 3, 7, 0, 12, 3, 59, 85, 86, 3, 66, 59, 86, 3, 260, 126, 116, 3, 127, 128, 32, 3, 6, 33, 34, 3, 133, 132, 136, 3, 108, 29, 35, 3, 29, 36, 35, 3, 36, 133, 136, 3, 144, 53, 11, 3, 242, 243, 58, 3, 224, 223, 190, 3, 153, 154, 65, 3, 193, 65, 26, 3, 24, 22, 18, 3, 22, 23, 18, 3, 18, 61, 68, 3, 67, 88, 102, 3, 178, 67, 102, 3, 102, 88, 89, 3, 157, 0, 158, 3, 3, 0, 157, 3, 78, 3, 157, 3, 23, 16, 61, 3, 19, 156, 155, 3, 19, 155, 62, 3, 25, 156, 19, 3, 128, 1, 2, 3, 1, 17, 2, 3, 32, 2, 81, 3, 32, 128, 2, 3, 2, 69, 81, 3, 2, 17, 69, 3, 49, 12, 47, 3, 47, 3, 48, 3, 47, 12, 3, 3, 77, 3, 78, 3, 77, 48, 3, 3, 82, 32, 81, 3, 85, 32, 82, 3, 91, 152, 148, 3, 91, 148, 96, 3, 91, 198, 152, 3, 27, 28, 111, 3, 111, 34, 38, 3, 28, 34, 111, 3, 111, 38, 110, 3, 243, 30, 58, 3, 30, 118, 4, 3, 58, 30, 4, 3, 118, 201, 202, 3, 126, 124, 116, 3, 4, 173, 174, 3, 58, 4, 57, 3, 57, 4, 174, 3, 118, 175, 4, 3, 118, 202, 175, 3, 4, 175, 173, 3, 218, 5, 6, 3, 6, 5, 33, 3, 5, 7, 33, 3, 158, 7, 5, 3, 218, 158, 5, 3, 131, 7, 12, 3, 33, 7, 131, 3, 136, 132, 138, 3, 138, 46, 139, 3, 138, 132, 46, 3, 108, 35, 145, 3, 35, 36, 207, 3, 36, 136, 207, 3, 29, 37, 36, 3, 110, 38, 37, 3, 110, 37, 29, 3, 9, 144, 11, 3, 8, 52, 53, 3, 144, 8, 53, 3, 144, 52, 8, 3, 143, 144, 9, 3, 143, 9, 11, 3, 123, 10, 11, 3, 41, 10, 123, 3, 143, 11, 10, 3, 41, 143, 10, 3, 40, 41, 123, 3, 210, 40, 121, 3, 55, 79, 44, 3, 204, 114, 70, 3, 70, 72, 170, 3, 113, 14, 114, 3, 11, 53, 14, 3, 125, 14, 113, 3, 123, 11, 14, 3, 123, 14, 125, 3, 135, 49, 46, 3, 132, 135, 46, 3, 135, 12, 49, 3, 131, 12, 135, 3, 13, 52, 144, 3, 13, 46, 52, 3, 139, 13, 144, 3, 139, 46, 13, 3, 54, 55, 44, 3, 14, 53, 54, 3, 14, 54, 114, 3, 51, 48, 77, 3, 55, 51, 79, 3, 51, 77, 79, 3, 242, 58, 188, 3, 68, 61, 161, 3, 223, 197, 190, 3, 223, 226, 197, 3, 67, 178, 177, 3, 25, 15, 156, 3, 15, 157, 156, 3, 78, 157, 15, 3, 75, 15, 25, 3, 75, 78, 15, 3, 23, 20, 16, 3, 20, 19, 62, 3, 20, 62, 61, 3, 16, 20, 61, 3, 17, 18, 163, 3, 17, 163, 69, 3, 163, 18, 68, 3, 227, 168, 231, 3, 170, 72, 171, 3, 204, 70, 203, 3, 167, 25, 169, 3, 169, 25, 19, 3, 169, 20, 23, 3, 169, 19, 20, 3, 21, 23, 22, 3, 21, 169, 23, 3, 168, 169, 21, 3, 24, 21, 22, 3, 231, 168, 21, 3, 228, 21, 24, 3, 228, 231, 21, 3, 167, 74, 25, 3, 74, 75, 25, 3, 44, 79, 76, 3, 44, 76, 72, 3, 85, 82, 84, 3, 178, 180, 185, 3, 96, 148, 147, 3, 213, 147, 212, 3, 95, 213, 248, 3, 95, 96, 147, 3, 95, 147, 213, 3, 97, 91, 96, 3, 189, 92, 93, 3, 181, 183, 182, 3, 181, 98, 183, 3, 178, 181, 180, 3, 93, 103, 101, 3, 103, 98, 181, 3, 164, 89, 165, 3, 102, 89, 164, 3, 101, 102, 164, 3, 105, 93, 101, 3, 189, 93, 105, 3, 105, 101, 164, 3, 198, 106, 152, 3, 198, 112, 106, 3, 189, 196, 92, 3, 196, 198, 91, 3, 196, 91, 92, 3, 193, 26, 192, 3, 192, 27, 111, 3, 192, 28, 27, 3, 192, 26, 28, 3, 198, 192, 112, 3, 112, 110, 107, 3, 107, 29, 108, 3, 107, 110, 29, 3, 243, 119, 30, 3, 260, 116, 119, 3, 260, 119, 243, 3, 119, 118, 30, 3, 125,  [... cut ...]
-0.11540962399721466092, 0.74508239352308680559, -0.65691159645018415425, -0.07581716522176343287,
-0.21483559984200240045, -0.66440074140450477991, -0.7158333045211513479, 0.039059250509023583919,
-0.96980734361487930251, 0.05851394434707302139, 0.23674846269321175862, 0.023432879780773328143,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000'
```

SeanCurtis-TRI, 2024-07-02:
I'd emphasize (from f2f conversation) that you're getting these with some frequency.

nepfaff, 2024-07-07:
I am but I'm also solving tons of these problems with different initial states. I'm running ~500 per script and around 15% of my scripts end up failing because of this.

hongkai-dai, 2024-07-08:
This happens when we call the EPA algorithm in FCL, but EPA thinks the two objects are separated. When the two objects are touching (with signed distance ~ 0), it is very hard to distinguish penetration (distance  0) versus separating (distance > 0).

I would propose that we modify FCL source code, that instead of throwing an error, just report that distance = 0 (and we need to back out the witness points as well).

cohnt, 2024-09-25:
I'm encountering this error as well, but in my case, it's with two boxes. Happy to create a separate issue, but figured I would comment here first. Output pasted below:
```
RuntimeError: Error with configuration
  Original error message: bazel-out/k8-opt/bin/external/fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Box<double>(0.050000000000000002776, 0.55000000000000004441, 0.2999999999999999889);
  X_FS1
0.00079632671073326442932, -0.99999968293183538748, 0, 0.75000000000000055511,
0.99999968293183538748, 0.00079632671073326442932, 0, -2.4083553715553102684e-15,
0, 0, 1, 0.14999999999999999445,
0, 0, 0, 1;
  Shape 2:
Box<double>(0.25, 0.2000000000000000111, 0.14999999999999999445);
  X_FS2
6.1232339957367660359e-17, 0, 1, 0.75,
-1, 6.1232339957367660359e-17, 6.1232339957367660359e-17, 0,
-6.1232339957367660359e-17, -1, 3.7493994566546440196e-33, 0.14999999999999999445,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```

nepfaff, 2024-09-27:
Got another full message. This time from `ComputeSignedDistancePairwiseClosestPoints`:

```
    ] = query_object.ComputeSignedDistancePairwiseClosestPoints(max_distance=-1e-12)
RuntimeError: Error with configuration
  Original error message: bazel-out/k8-opt/bin/external/fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.094311179999999994461, 0.039064990000000000825, 0.013511900000000000299),
      Vector3<double>(0.040435829999999998996, -0.097620650000000003255, 0.016015359999999999319),
      Vector3<double>(0.097620670000000006605, -0.040435739999999997796, 0.016015359999999999319),
      Vector3<double>(0.094311199999999997812, -0.039064929999999997712, 0.013511900000000000299),
      Vector3<double>(-0.040616260000000001251, -0.098056500000000004769, 0.017358849999999998587),
      Vector3<double>(0.098056459999999998067, 0.040616359999999997188, 0.017358849999999998587),
      Vector3<double>(0.09762063999999999464, 0.040435829999999998996, 0.016015359999999999319),
      Vector3<double>(0.039064960000000002738, 0.094311199999999997812, 0.013511900000000000299),
      Vector3<double>(-0.040435760000000001146, -0.097620680000000001342, 0.016015359999999999319),
      Vector3<double>(-0.039064939999999999387, -0.094311210000000006426, 0.013511900000000000299),
      Vector3<double>(-0.098056459999999998067, -0.040616369999999998863, 0.017358849999999998587),
      Vector3<double>(0.039064990000000000825, -0.094311190000000003075, 0.013511900000000000299),
      Vector3<double>(0.097182140000000000191, -0.040254119999999997404, 0.017967910000000000131),
      Vector3<double>(0.040254159999999997166, -0.097182130000000005454, 0.017967910000000000131),
      Vector3<double>(0.09718211999999999684, 0.040254169999999998841, 0.017967910000000000131),
      Vector3<double>(0.098056509999999999505, -0.040616260000000001251, 0.017358849999999998587),
      Vector3<double>(0.040616359999999997188, -0.098056470000000006682, 0.017358849999999998587),
      Vector3<double>(-0.040254119999999997404, -0.097182140000000000191, 0.017967910000000000131),
      Vector3<double>(-0.09718211999999999684, -0.040254180000000000517, 0.017967910000000000131),
      Vector3<double>(0.04025415000000000243, 0.097182140000000000191, 0.017967910000000000131),
      Vector3<double>(-0.027032799999999999191, 0.065262920000000002041, 0.00021531999999999999165),
      Vector3<double>(0.027032770000000001104, 0.065262929999999996777, 0.00021531999999999999165),
      Vector3<double>(-0.065262920000000002041, 0.027032770000000001104, 0.00021531999999999999165),
      Vector3<double>(0.027032799999999999191, -0.065262920000000002041, 0.00021531999999999999165),
      Vector3<double>(0.065262940000000005392, -0.027032750000000001223, 0.00021531999999999999165),
      Vector3<double>(-0.027032759999999999428, -0.065262940000000005392, 0.00021531999999999999165),
      Vector3<double>(0.065262909999999993427, 0.027032799999999999191, 0.00021531999999999999165),
      Vector3<double>(-0.065262909999999993427, -0.027032810000000000866, 0.00021531999999999999165),
      Vector3<double>(-0.039064990000000000825, 0.094311190000000003075, 0.013511900000000000299),
      Vector3<double>(0.040435770000000002822, 0.097620670000000006605, 0.016015359999999999319),
      Vector3<double>(-0.097620659999999997991, 0.040435770000000002822, 0.016015359999999999319),
      Vector3<double>(-0.094311179999999994461, -0.0390650000000000025, 0.013511900000000000299),
      Vector3<double>(-0.094311190000000003075, 0.039064960000000002738, 0.013511900000000000299),
      Vector3<double>(-0.09762063999999999464, -0.040435840000000000671, 0.016015359999999999319),
      Vector3<double>(-0.040616359999999997188, 0.098056470000000006682, 0.017358849999999998587),
      Vector3<double>(-0.040435829999999998996, 0.097620650000000003255, 0.016015359999999999319),
      Vector3<double>(-0.098056489999999996154, 0.040616289999999999338, 0.017358849999999998587),
      Vector3<double>(-0.09718211999999999684, 0.04025415000000000243, 0.017967910000000000131),
      Vector3<double>(-0.040254169999999998841, 0.097182130000000005454, 0.017967910000000000131),
      Vector3<double>(0.040616289999999999338, 0.098056500000000004769, 0.017358849999999998587),}),
    66,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 26, 0, 24, 3, 9, 25, 11, 3, 38, 19, 39, 3, 10, 18, 36, 3, 2, 0, 6, 3, 15, 2, 6, 3, 24, 0, 3, 3, 3, 0, 2, 3, 11, 24, 3, 3, 6, 0, 7, 3, 7, 26, 21, 3, 7, 0, 26, 3, 10, 8, 4, 3, 8, 9, 11, 3, 4, 8, 16, 3, 31, 27, 25, 3, 31, 25, 9, 3, 22, 27, 31, 3, 32, 22, 31, 3, 16, 1, 15, 3, 1, 2, 15, 3, 8, 1, 16, 3, 8, 11, 1, 3, 1, 3, 2, 3, 1, 11, 3, 3, 16, 13, 4, 3, 4, 17, 10, 3, 17, 18, 10, 3, 13, 17, 4, 3, 18, 37, 36, 3, 5, 14, 15, 3, 15, 6, 5, 3, 5, 6, 39, 3, 19, 14, 5, 3, 19, 5, 39, 3, 6, 29, 39, 3, 6, 7, 29, 3, 7, 21, 28, 3, 29, 7, 28, 3, 33, 9, 8, 3, 33, 31, 9, 3, 10, 33, 8, 3, 36, 33, 10, 3, 23, 24, 11, 3, 25, 23, 11, 3, 15, 14, 12, 3, 15, 12, 16, 3, 12, 13, 16, 8, 18, 17, 13, 12, 14, 19, 38, 37, 3, 28, 21, 20, 3, 20, 22, 32, 3, 28, 20, 32, 8, 22, 20, 21, 26, 24, 23, 25, 27, 3, 35, 28, 32, 3, 29, 28, [... cut ...] 3, 3, 36, 30, 33, 3, 35, 30, 36, 3, 35, 32, 30, 3, 34, 35, 36, 3, 37, 38, 34, 3, 36, 37, 34, 3, 34, 38, 39, 3, 39, 35, 34,}),
    false);
  X_FS2
0.99999999967614783269, 2.5450037351646537633e-05, 0, 0,
-2.5450037351646537633e-05, 0.99999999967614783269, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
"""
```

nepfaff, 2024-12-09:
Is it still useful to post more examples of these? I initially understood that they are printed in a way that they can be directly added as test cases, but then they are probably all due to the same issue. I'm getting a couple dozen of these a day at the moment, so I'd be happy to collect some if helpful.

cohnt, 2025-02-25:
Awesome awesome awesome! This issue has been a major problem for me on multiple projects now, so thanks for getting that fix in!

cohnt, 2025-02-27:
Here's another instance (this one's a hexagonal prism), in case it helps you:
```
  Original error message: bazel-out/k8-opt/bin/external/drake++internal_repositories+fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.5, 0.5, -0.86602500000000004476),
      Vector3<double>(0.5, -0.5, -0.86602500000000004476),
      Vector3<double>(0.5, 1, 0),
      Vector3<double>(-0.5, -0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, 0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, -1, 0),
      Vector3<double>(0.5, -0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, 0.5, -0.86602500000000004476),
      Vector3<double>(-0.5, -0.5, -0.86602500000000004476),
      Vector3<double>(-0.5, 1, 0),
      Vector3<double>(0.5, -1, 0),
      Vector3<double>(0.5, 0.5, 0.86602500000000004476),}),
    8,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             4, 10, 5, 8, 1, 6, 2, 11, 6, 10, 1, 0, 4, 0, 7, 9, 2, 4, 1, 8, 7, 0, 4, 2, 9, 4, 11, 6, 8, 5, 3, 4, 9, 7, 4, 6, 3, 5, 10, 4, 4, 3, 6, 11,}),
    false);
  X_FS1
-3.0140487242790098936e-09, 0.34009658459984087875, -0.94039051098122172778, 4.2864836941313040342,
1.090044683538143324e-09, 0.94039051098122172778, 0.34009658459984087875, 7.3869576872253679412,
1, 0, -3.2051032938795742666e-09, 0,
0, 0, 0, 1;
  Shape 2:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(4.1494020000000002568, 8.4937090000000008416, -0.5),
      Vector3<double>(5.1884509999999997021, 7.476263000000000325, -0.5),
      Vector3<double>(3.9316469999999998919, 7.3745029999999998083, -0.5),
      Vector3<double>(4.1494020000000002568, 8.4937090000000008416, 0.5),
      Vector3<double>(3.9316469999999998919, 7.3745029999999998083, 0.5),
      Vector3<double>(5.1884509999999997021, 7.476263000000000325, 0.5),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 5, 3, 4, 4, 2, 1, 5, 4, 3, 2, 0, 1, 4, 3, 0, 2, 4, 4, 1, 0, 3, 5,}),
    false);
  X_FS2
1, 0, 0, 0,
0, 1, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```

cohnt, 2025-02-27:
And here's a heptagonal prism:
```
  Original error message: bazel-out/k8-opt/bin/external/drake++internal_repositories+fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.5, 1, 0),
      Vector3<double>(0.5, 0.62348999999999998867, -0.7818310000000000537),
      Vector3<double>(0.5, -0.22252099999999999658, -0.97492800000000001681),
      Vector3<double>(0.5, 0.62348999999999998867, 0.7818310000000000537),
      Vector3<double>(-0.5, -0.90096900000000001985, 0.4338839999999999919),
      Vector3<double>(-0.5, -0.22252099999999999658, 0.97492800000000001681),
      Vector3<double>(-0.5, -0.90096900000000001985, -0.4338839999999999919),
      Vector3<double>(0.5, -0.90096900000000001985, 0.4338839999999999919),
      Vector3<double>(-0.5, 0.62348999999999998867, -0.7818310000000000537),
      Vector3<double>(-0.5, 0.62348999999999998867, 0.7818310000000000537),
      Vector3<double>(-0.5, -0.22252099999999999658, -0.97492800000000001681),
      Vector3<double>(-0.5, 1, 0),
      Vector3<double>(0.5, -0.90096900000000001985, -0.4338839999999999919),
      Vector3<double>(0.5, -0.22252099999999999658, 0.97492800000000001681),}),
    9,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             4, 10, 2, 12, 6, 4, 1, 8, 11, 0, 4, 10, 8, 1, 2, 7, 2, 1, 0, 3, 13, 7, 12, 4, 9, 3, 0, 11, 4, 9, 5, 13, 3, 7, 8, 10, 6, 4, 5, 9, 11, 4, 12, 7, 4, 6, 4, 13, 5, 4, 7,}),
    false);
  X_FS1
-2.0479437360480804916e-09, -0.76923712747984118732, -0.63896341186844385351, 13.670417904867957049,
-2.4654844510601008403e-09, 0.63896341186844385351, -0.76923712747984118732, 13.169816779836704512,
1, 0, -3.2051032938795742666e-09, 0,
0, 0, 0, 1;
  Shape 2:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(14.835100999999999871, 13.060409999999999187, -0.5),
      Vector3<double>(13.785037000000000873, 12.890523999999999205, -0.5),
      Vector3<double>(14.767995000000000871, 13.150247999999999493, -0.5),
      Vector3<double>(14.835100999999999871, 13.060409999999999187, 0.5),
      Vector3<double>(14.767995000000000871, 13.150247999999999493, 0.5),
      Vector3<double>(13.785037000000000873, 12.890523999999999205, 0.5),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 3, 4, 5, 4, 2, 1, 5, 4, 3, 2, 0, 1, 4, 3, 0, 2, 4, 4, 5, 1, 0, 3,}),
    false);
  X_FS2
1, 0, 0, 0,
0, 1, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```

cohnt, 2025-02-27:
Here's a rectangular prism causing a problem:
```
  Original error message: bazel-out/k8-opt/bin/external/drake++internal_repositories+fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.5, -0.92388000000000003453, 0.38268299999999999539),
      Vector3<double>(-0.5, 0.92388000000000003453, -0.38268299999999999539),
      Vector3<double>(0.5, 0.92388000000000003453, -0.38268299999999999539),
      Vector3<double>(-0.5, 0.92388000000000003453, 0.38268299999999999539),
      Vector3<double>(-0.5, -0.92388000000000003453, -0.38268299999999999539),
      Vector3<double>(0.5, 0.92388000000000003453, 0.38268299999999999539),
      Vector3<double>(-0.5, -0.92388000000000003453, 0.38268299999999999539),
      Vector3<double>(0.5, -0.92388000000000003453, -0.38268299999999999539),}),
    6,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             4, 3, 6, 0, 5, 4, 7, 2, 5, 0, 4, 6, 4, 7, 0, 4, 2, 1, 3, 5, 4, 3, 1, 4, 6, 4, 7, 4, 1, 2,}),
    false);
  X_FS1
-2.6487935924268804787e-09, -0.56303944378188575115, -0.82643002410717436579, 15.712812998638135298,
-1.8045995758494454423e-09, 0.82643002410717436579, -0.56303944378188575115, 10.219809745800642276,
1, 0, -3.2051032938795742666e-09, 0,
0, 0, 0, 1;
  Shape 2:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(15.417082000000000619, 9.6986209999999992704, -0.5),
      Vector3<double>(16.236605000000000842, 9.2332309999999999661, -0.5),
      Vector3<double>(16.224319999999998743, 8.9158489999999996911, -0.5),
      Vector3<double>(15.417082000000000619, 9.6986209999999992704, 0.5),
      Vector3<double>(16.224319999999998743, 8.9158489999999996911, 0.5),
      Vector3<double>(16.236605000000000842, 9.2332309999999999661, 0.5),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 3, 4, 5, 4, 2, 1, 5, 4, 3, 2, 0, 1, 4, 4, 3, 0, 2, 4, 5, 1, 0, 3,}),
    false);
  X_FS2
1, 0, 0, 0,
0, 1, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```

cohnt, 2025-02-27:
Another instance
```
  Original error message: bazel-out/k8-opt/bin/external/drake++internal_repositories+fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.5, 0.5, -0.86602500000000004476),
      Vector3<double>(0.5, -0.5, -0.86602500000000004476),
      Vector3<double>(0.5, 1, 0),
      Vector3<double>(-0.5, -0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, 0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, -1, 0),
      Vector3<double>(0.5, -0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, 0.5, -0.86602500000000004476),
      Vector3<double>(-0.5, -0.5, -0.86602500000000004476),
      Vector3<double>(-0.5, 1, 0),
      Vector3<double>(0.5, -1, 0),
      Vector3<double>(0.5, 0.5, 0.86602500000000004476),}),
    8,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             4, 10, 5, 8, 1, 6, 2, 11, 6, 10, 1, 0, 4, 0, 7, 9, 2, 4, 1, 8, 7, 0, 4, 2, 9, 4, 11, 6, 8, 5, 3, 4, 9, 7, 4, 6, 3, 5, 10, 4, 4, 3, 6, 11,}),
    false);
  X_FS1
-2.2411796174427631456e-09, 0.7148738189791493669, -0.69925347545662319693, 6.3335747278141161232,
2.2912444319183419853e-09, 0.69925347545662319693, 0.7148738189791493669, 8.4631576303276716544,
1, 0, -3.2051032938795742666e-09, 0,
0, 0, 0, 1;
  Shape 2:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(5.2423630000000001061, 9.1228110000000004476, -0.5),
      Vector3<double>(6.3596640000000004278, 8.2852599999999991809, -0.5),
      Vector3<double>(5.6703770000000002227, 7.5985389999999997102, -0.5),
      Vector3<double>(5.2423630000000001061, 9.1228110000000004476, 0.5),
      Vector3<double>(5.6703770000000002227, 7.5985389999999997102, 0.5),
      Vector3<double>(6.3596640000000004278, 8.2852599999999991809, 0.5),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 3, 4, 5, 4, 4, 2, 1, 5, 3, 2, 0, 1, 4, 3, 0, 2, 4, 4, 1, 0, 3, 5,}),
    false);
  X_FS2
1, 0, 0, 0,
0, 1, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```

cohnt, 2025-02-28:
Some more instances
```
  Original error message: bazel-out/k8-opt/bin/external/drake++internal_repositories+fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(-0.5, -0.5, -0.86602500000000004476),
      Vector3<double>(-0.5, -0.5, 0.86602500000000004476),
      Vector3<double>(-0.5, 1, 0),
      Vector3<double>(0.5, -0.5, -0.86602500000000004476),
      Vector3<double>(0.5, 1, 0),
      Vector3<double>(0.5, -0.5, 0.86602500000000004476),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 5, 3, 4, 4, 2, 1, 5, 4, 3, 2, 0, 1, 4, 3, 0, 2, 4, 4, 1, 0, 3, 5,}),
    false);
  X_FS1
-3.155367699156126597e-09, -0.17548349096519452739, -0.98448237383849002136, 9.4277544959429171456,
-5.6244271491403150214e-10, 0.98448237383849002136, -0.17548349096519452739, 4.3549730982604870633,
1, 0, -3.2051032938795742666e-09, 0,
0, 0, 0, 1;
  Shape 2:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(7.502455000000000318, 3.9926669999999999661, -0.5),
      Vector3<double>(7.9127499999999999503, 2.7334600000000000009, -0.5),
      Vector3<double>(8.6743819999999995929, 3.7108469999999997846, -0.5),
      Vector3<double>(7.502455000000000318, 3.9926669999999999661, 0.5),
      Vector3<double>(8.6743819999999995929, 3.7108469999999997846, 0.5),
      Vector3<double>(7.9127499999999999503, 2.7334600000000000009, 0.5),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 3, 5, 4, 4, 5, 1, 2, 4, 3, 1, 0, 2, 4, 4, 2, 0, 3, 4, 3, 0, 1, 5,}),
    false);
  X_FS2
1, 0, 0, 0,
0, 1, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```
```
  Original error message: bazel-out/k8-opt/bin/external/drake++internal_repositories+fcl_internal/drake_hdr/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1720): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(0.5, 0.30901699999999998614, -0.95105700000000004124),
      Vector3<double>(0.5, 1, 0),
      Vector3<double>(-0.5, -0.80901699999999998614, 0.58778500000000000192),
      Vector3<double>(-0.5, 0.30901699999999998614, 0.95105700000000004124),
      Vector3<double>(-0.5, -0.80901699999999998614, -0.58778500000000000192),
      Vector3<double>(0.5, -0.80901699999999998614, 0.58778500000000000192),
      Vector3<double>(-0.5, 0.30901699999999998614, -0.95105700000000004124),
      Vector3<double>(-0.5, 1, 0),
      Vector3<double>(0.5, -0.80901699999999998614, -0.58778500000000000192),
      Vector3<double>(0.5, 0.30901699999999998614, 0.95105700000000004124),}),
    7,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             4, 0, 6, 7, 1, 5, 9, 5, 8, 0, 1, 4, 0, 8, 4, 6, 4, 1, 7, 3, 9, 5, 6, 4, 2, 3, 7, 4, 8, 5, 2, 4, 4, 3, 2, 5, 9,}),
    false);
  X_FS1
-3.1934130613594990691e-09, -0.085331461972016672823, -0.99635261910516315087, 11.261494468063983021,
-2.7349614983807029573e-10, 0.99635261910516315087, -0.085331461972016672823, 5.7102314848013024928,
1, 0, -3.2051032938795742666e-09, 0,
0, 0, 0, 1;
  Shape 2:
Convex<double>(
  std::make_shared<std::vector<Vector3<double>>>(
    std::initializer_list<Vector3<double>>{
      Vector3<double>(10.924768999999999508, 5.6701019999999999754, -0.5),
      Vector3<double>(10.83925699999999992, 4.8329899999999996751, -0.5),
      Vector3<double>(11.452507000000000659, 5.5209809999999999164, -0.5),
      Vector3<double>(10.924768999999999508, 5.6701019999999999754, 0.5),
      Vector3<double>(11.452507000000000659, 5.5209809999999999164, 0.5),
      Vector3<double>(10.83925699999999992, 4.8329899999999996751, 0.5),}),
    5,
    std::make_shared<std::vector<int>>(
        std::initializer_list<int>{
             3, 4, 3, 5, 4, 5, 1, 2, 4, 3, 1, 0, 2, 4, 2, 0, 3, 4, 4, 5, 3, 0, 1,}),
    false);
  X_FS2
1, 0, 0, 0,
0, 1, 0, 0,
0, 0, 1, 0,
0, 0, 0, 1;
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```