RobotLocomotion/drake #21675: BoxBoxGradient Failure in KinematicTrajectoryOptimization

cohnt, 2024-07-02:
### What happened?

When solving a KinematicTrajectoryOptimization problem with a minimum distance constraint, and box geometries on both the "robot" and world, I'm consistently getting the following failure when I try to solve the problem:
```
Traceback (most recent call last):
  File "...", line 301, in <module>
    result = Solve(trajopt.prog())
RuntimeError: Exception while evaluating SNOPT costs and constraints: 'Failure at geometry/proximity/distance_to_shape_touching.cc:123 in BoxBoxGradient(): condition '(X_WA * p_ACa - X_WB * p_BCb).norm() < kEps' failed.'
```
Looks like that norm ends up being larger than `1e-14`, which is the value used for `kEps`  in [BoxBoxGradient](https://github.com/RobotLocomotion/drake/blob/af0551e35fbe97bcb9f0e983945002ecbe4e15f0/geometry/proximity/distance_to_shape_touching.cc#L121). But shouldn't this be caught by the check in [`CalcDistanceFallback`](https://github.com/RobotLocomotion/drake/blob/af0551e35fbe97bcb9f0e983945002ecbe4e15f0/geometry/proximity/distance_to_shape_callback.cc#L92), which uses the same `kEps`?

I'm able to consistently replicate the bug with [this gist](https://gist.github.com/cohnt/78069217f56ca3e71db6dd738027e75b) on master -- you'll just have to unzip the models folder to run it. Unfortunately, I haven't been able to distill a simpler example. And I'm not familiar enough with Drake's collision engine to debug further.

### Version

af0551e35f

RussTedrake, 2024-07-03:
+@SeanCurtis-TRI for triage.

juanmed, 2025-11-30:
I am experiencing this issue as well. My simulation consist of 3 boxes colliding (red, yellow and gray box -floor- ). The issue is with the contact between the yellow and gray (floor) boxes, as when I add a z dof to the yellow box, the issue goes away. 

https://github.com/user-attachments/assets/e71fe9cf-6629-485b-9cb0-9e2a9492abaa

```
abort: Failure at external/drake+/geometry/proximity/distance_to_shape_touching.cc:122 in BoxBoxGradient(): condition '(X_WA * p_ACa - X_WB * p_BCb).norm() < kEps' failed.
```
The following are my sdf files (attached as .txt because github would complain .sdf is not supported) from which I created a multibody plant and simply move the red box towards the yellow one. 

Any ideas how to handle this? 

[conveyor_belt_tool_1d.txt](https://github.com/user-attachments/files/23840290/conveyor_belt_tool_1d.txt)

[box.txt](https://github.com/user-attachments/files/23840292/box.txt)

cohnt, 2026-10-27:
This still reproduces on master (e846fe2316). Here's a minimal reproduction with two hardcoded boxes, taken from a configuration where the trajectory optimization failed:

```python
import numpy as np
from pydrake.geometry import (
    Box, FramePoseVector, GeometryFrame, GeometryInstance, ProximityProperties,
    SceneGraph)
from pydrake.math import RigidTransform, RotationMatrix

scene_graph = SceneGraph()
source_id = scene_graph.RegisterSource("source")
frame_A = scene_graph.RegisterFrame(source_id, GeometryFrame("A"))
frame_B = scene_graph.RegisterFrame(source_id, GeometryFrame("B"))
for frame, box in [(frame_A, Box(0.05, 0.55, 0.3)), (frame_B, Box(0.4, 0.3, 0.2))]:
    geometry_id = scene_graph.RegisterGeometry(
        source_id, frame, GeometryInstance(RigidTransform(), box, "box"))
    scene_graph.AssignRole(source_id, geometry_id, ProximityProperties())

X_WA = RigidTransform(
    RotationMatrix(np.array([
        [1.0000000000000044, 2.2886701630728595e-19, 0.0],
        [1.0581860424029641e-20, 1.0000000000000044, 0.0],
        [0.0, 0.0, 1.0]])),
    [3.5000000000000107, 2.2500000000000027, 0.15])
X_WB = RigidTransform(
    RotationMatrix(np.array([
        [1.0, 0.0, 0.0],
        [0.0, -0.16393107388774433, 0.9864717953464309],
        [0.0, -0.9864717953464309, -0.16393107388774433]])),
    [3.2750000000000203, 2.6252702917956396, 0.16073032646376256])

context = scene_graph.CreateDefaultContext()
poses = FramePoseVector()
poses.set_value(frame_A, X_WA)
poses.set_value(frame_B, X_WB)
scene_graph.get_source_pose_port(source_id).FixValue(context, poses)
query_object = scene_graph.get_query_output_port().Eval(context)
print(query_object.ComputeSignedDistancePairwiseClosestPoints())
```

```
Failure at geometry/proximity/distance_to_shape_touching.cc:122 in BoxBoxGradient(): condition '(X_WA * p_ACa - X_WB * p_BCb).norm() < kEps' failed.
```

The boxes penetrate very slightly. FCL reports a signed distance of -9.77e-15, so `CalcDistanceFallback()` treats them as touching (|d| < 1e-14) and calls `BoxBoxGradient()`. The witness points are legitimately |d| = 9.77e-15 apart, and transforming them into each box's frame and back to world adds enough rounding (the boxes are ~3.5 m from the origin) to make the gap 1.06e-14, which fails the `DRAKE_DEMAND`. The two checks use the same tolerance, so any penetration close to 1e-14 leaves no margin for rounding.

closed 2026-11-04 by SeanCurtis-TRI