RobotLocomotion/drake #21682: Non-Repeatability in KinematicTrajectoryOptimization

cohnt, 2024-07-03:
### What happened?

Per the discussion in [this stackoverflow post](https://stackoverflow.com/questions/77363144/direct-collocation-determinism) and a discussion with @hongkai-dai, I've been able to construct a trajopt instance where MathematicalProgram has nondeterministic behavior. It can be found [here](https://gist.github.com/cohnt/632072078cd4b451e89c095edc2db427) -- if one examines the SNOPT logs, they clearly diverge at some point during the solve. I'm suspicious of the MinimumDistanceLowerBoundConstraint, given that the logs are identical if that constraint is removed.

hongkai-dai, 2024-07-04:
@cohnt in your python script, it looks for `package://se3maze/models/maze.sdf` and `package://se3maze/models/box.sdf`. I don't have these two sdf files. Could you share the sdf files as well? Thanks!

cohnt, 2024-07-04:
A zip file with the models and a package.xml are also included in the gist. You'll just have to unzip it to get the files. (I think gist doesn't allow you to use directories.)

hongkai-dai, 2024-07-06:
Sorry it took a while. I have pinned down to the function [QueryObject::ComputeSignedDistancePairwiseClosestPoints](https://github.com/RobotLocomotion/drake/blob/9f5a09ca7621a35450fb39905a28725d9eb69b8a/geometry/query_object.h#L664) which return a `std::vector<SignedDistancePair<T>>` object. For the same input, the returned `std::vector` should have the same ordering, but in reality the ordering changes.

Here is the computed distances in one call
```
distances[0]=0.05217582339447376233, grad[25]=0.00000000000000056585, grad[31]=0.00000000000000044963
distances[1]=0.07077759825297903762, grad[25]=-0.35937297264284862042, grad[31]=-0.28555681000760274602
distances[2]=0.04804141069317556523, grad[25]=-0.53185542592967094411, grad[31]=-0.42261090948719659544
```
Here is the output in another call
```
distances[0]=0.04804141069317556523, grad[25]=-0.53185542592967094411, grad[31]=-0.42261090948719659544
distances[1]=0.05217582339447376233, grad[25]=0.00000000000000056585, grad[31]=0.00000000000000044963
distances[2]=0.07077759825297903762, grad[25]=-0.35937297264284862042, grad[31]=-0.28555681000760274602
```

I will open a separate github issue to track this (since the problem is not in KinematicTrajectoryOptimization, but in evaluating signed distance in our collision engine).

jwnimmer-tri, 2024-07-06:
To be sure I'm following... the distance returned are always the same bit-exact set, but just in a different order?

We eventually pass those distances to `math::SoftOverMax` to fold.  With infinite-precision real numbers, input order would not affect its output, but with floating point round-off during the summation, the input order of the distances can affect the some low-order bits of the result?

hongkai-dai, 2024-07-06:
Yes, the distance is the same bit-exact set, just the ordering is different, and `math::SoftOverMax` can generate slightly different result (differ by 1E-15) due to the ordering change.

cohnt, 2024-07-12:
Confirmed that merging #21698 fixes this issue.

closed 2024-07-12 by cohnt