RobotLocomotion/drake #21708: [planning] Adding large numbers of collision geometries to SceneGraphCollisionChecker is slow

calderpg-tri, 2024-07-11:
### What happened?

When [adding collision geometry](https://github.com/RobotLocomotion/drake/blob/master/planning/scene_graph_collision_checker.cc#L61) to `SceneGraphCollisionChecker`, [ApplyCollisionFiltersToSceneGraph](https://github.com/RobotLocomotion/drake/blob/master/planning/scene_graph_collision_checker.cc#L366) is called after the geometry is added. If the user is adding a large number of geometries at once, this results in the filter update being computed and applied after each geometry, rather than once at the end of the batch.

We should modify the existing batch-addition API in `CollisionChecker` to run the update only once per batch.

cc @tomstewart-woven @aochiai-woven

calderpg-tri, 2024-07-11:
fyi @jwnimmer-tri

jwnimmer-tri, 2024-07-15:
On clarification -- the public API already is plural (multiple shape), it's just that the base class NVI spoon feeds them one by one do the base class.