RobotLocomotion/drake #21715: Jupyter tests fail on Mac Ventura CI

BetsyMcPhail, 2024-07-15:
Initial report in slack: https://drakedevelopers.slack.com/archives/C270MN28G/p1721052417140899

List of failures:

- 7/12 [mac-arm-ventura-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-ventura-clang-bazel-continuous-release/658/)
- 7/13 [mac-arm-ventura-clang-bazel-nightly-debug](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-debug/177/)
- 7/14 [mac-arm-ventura-clang-bazel-nightly-everything-release](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-everything-release/433/)
- 7/15 [mac-arm-ventura-clang-bazel-nightly-debug](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-debug/179/)
- 7/15 [mac-arm-ventura-clang-bazel-experimental-release](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/mac-arm-ventura-clang-bazel-experimental-release/7000/)

The failing tests vary, but the error output consistently contains:

```
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.12/site-packages/traitlets/traitlets.py", line 632, in get
    value = obj._trait_values[self.name]
            ~~~~~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'session'
```

BetsyMcPhail, 2024-07-15:
7/15 - [mac-arm-ventura-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-ventura-clang-bazel-continuous-release/664/)

jwnimmer-tri, 2024-07-15:
FYI to buildcops -- to the extent these tests are causing undue effort for the cops, we should disable them (on macOS CI).  This is doubly true if the error are due to packages out of our control, e.g., broken Homebrew / pip stuff.

For now I don't plan to take action, but please speak up if it's causing pain, either for cops, or for PR authors.  (For PR authors, we can remind them that macOS is not required pre-merge.)

jwnimmer-tri, 2024-07-16:
Also seen in my PR: https://drake-jenkins.csail.mit.edu/job/mac-arm-ventura-clang-bazel-experimental-release/7031/consoleText

BetsyMcPhail, 2024-07-17:
7/17 - https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-everything-release/437/

BetsyMcPhail, 2024-07-17:
7/17 - https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-ventura-clang-bazel-continuous-release/672/

BetsyMcPhail, 2024-07-25:
This error hasn't appeared since the Mac image was reprovisioned on 7/17, closing

closed 2024-07-25 by BetsyMcPhail