RobotLocomotion/drake #21725: Add a constructor to `DirectCollocation` and `DirectTranscription` that does not call `System<double>::ToAutoDiffXd(.)`

samzapo, 2024-07-17:
**Is your feature request related to a problem? Please describe.**
The following code accepts a `System<double>` and `Context<double>` then immediate converts them to use `AutoDiffXd`.
https://github.com/RobotLocomotion/drake/blob/84f4fc58557043bb73bb3ae2d2748c62b054ec4f/planning/trajectory_optimization/direct_collocation.cc#L236-L257

Implementing the various requirements necessary to call `System::ToAutoDiffXd()` is significantly more difficult than the caller simply constructing an `AutoDiffXd` system and its context themselves and passing it to the constructor.
 
**Describe the solution you'd like**
Add a constructor to `DirectCollocation` and `DirectTranscription` that accepts `double` and `AutoDiffXd` instantiations of the `System` and `Context`.

hongkai-dai, 2024-07-29:
Sorry for my belated reply. I was travelling last week without my programming computer.

I don't see a problem to add a constructor in `DirectCollocation` and `DirectTranscription` that accepts `AutoDiffXd` version `System<AutoDiffXd>` and `Context<AutoDiffXd>`. @RussTedrake as the original author of `DirectCollocation` class, are you OK with the additional constructors?

RussTedrake, 2024-08-19:
Sorry for _my_ delay.  Yes, we discussed this during the drake developers meeting. I think it seems very reasonable.