RobotLocomotion/drake #21756: HPolyhedron::MaximumVolumeInscribedEllipsoid() terminates program when applied to small polyhedra

Akshay5312, 2024-07-30:
### What happened?

Programs using HPolyhedron::MaximumVolumeInscribedEllipsoid() occasionally terminate with message 
```
thread '<unnamed>' panicked at external/crate__clarabel-0.6.0/src/solver/core/cones/psdtrianglecone.rs:292:9:
not implemented: Mixed PSD and Exponential/Power cones are not yet supported
```
when applied to small volume polyhedra. This was observed with a variant of IRIS-NP and might be related to #20839 , though the example polyhedron there just seemed ill scaled. An example failing polyhedron is documented [here](https://github.com/Akshay5312/Maximum_Volume_Inscribed_Ellipsoid).

The error message seems odd, given that the inscribed ellipsoid problem (with a log-det cost) inherently requires PSD and Exponential cone constraints, but the clarabel solver only fails sometimes. Using the SCS solver on the same polyhedron produced an inaccurate result. 

The radius of the contained chebyshev ball (for the example failing polyhedron) is on the order of 1e-5. One possible quick fix (explored [here](https://github.com/Akshay5312/drake_largestInscribedEllipsoidFix/blob/1f43f9f95e89b2a7e1346813de54934177e4926b/geometry/optimization/hpolyhedron.cc#L401)) is to scale and transform the polyhedron based on the chebyshev ball a-la 

```
 transformed_b = (b_ - A_ * chebyshev_center) / chebyshev_radius;
```
  
  

### Version

074d751b155244cb9cb4577bf111bb70afc71279 

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using CMake

### Relevant log output

```shell
thread '<unnamed>' panicked at external/crate__clarabel-0.6.0/src/solver/core/cones/psdtrianglecone.rs:292:9:
not implemented: Mixed PSD and Exponential/Power cones are not yet supported
```

jwnimmer-tri, 2024-07-31:
Does the bug also happen with the latest Drake?  I believe this was fixed as of v1.29.0 with the upgrade to Clarabel 0.7.1.

Akshay5312, 2024-07-31:
It seems the commit I had used pre-dates v1.29.0 by a month! That may have fixed it.

Akshay5312, 2024-07-31:
With the v 1.31.0 precompiledbinaries I get the error- 
```
[2024-07-30 20:45:43.900] [console] [info] Clarabel returns InsufficientProgress
terminate called after throwing an instance of 'std::runtime_error'
  what():  Solver Clarabel failed to solve the maximum inscribed ellipse problem; it terminated with SolutionResult SolverSpecificError). Make sure that your polyhedron is bounded and has an interior.
```

jwnimmer-tri, 2024-07-31:
Okay, well at least a catchable exception is better than a panic!

I'll ping other people more familiar with the mathematical formulation about the small polyhedra question.

Akshay5312, 2024-07-31:
Oh! I made a mistake! The test case that it failed for was not the one documented [here](https://github.com/Akshay5312/Maximum_Volume_Inscribed_Ellipsoid), it actually succeeded for that case! It failed for a box with ambient_dimension()==3, and with a side length of 2e-23.

AlexandreAmice, 2024-07-31:
Why exactly are you interested in solving the MaximumVolumeInscribedEllispoid program with such a small polytope? 

Considering that double precision machine epsilon is `1e-16`, you should not expect a numerical subroutine like solving a convex optimization program to perform very robustly when the data is already so poorly scaled.

For example, the polytope you have given is expressed in drake as:
`[I; -I] x <= 2e-23*1`. If I perturb any of the coefficients of this polytope by machine epsilon, the answer MaximumVolumeInscribedEllispoid changes from a ball centered at (0,0,0) with radius 1e-23 to infeasible. The typical precision of a semidefinite programming solver is usually only about the square root of machine epsilon, so the precision is about `1e-8` and so even if the solver did give you an answer to this problem, you probably shouldn't trust it. Therefore, the solver reporting an error is the right answer. 

If you really need to solve this type of problem, consider rescaling your problem data to different units (e.g. meters to zeptometers) before calling the numerical subroutines.

closed 2024-07-31 by AlexandreAmice