RobotLocomotion/drake #21778: [Meshcat] Geometries with paths prefixed by `/drake/illustration` aren't set

siddancha, 2024-08-05:
### What happened?

## Issue

I'm trying to visualize a geometry at a fixed transform from one of the robot links. To do so, I'm calling `Meshcat::SetObject()` with the path `/drake/illustration/iiwa/iiwa_link_7/red_ball`. However, it seems that any path that is prefixed by `/drake/illustration` or  `/drake/proximity` or `/drake/inertia` fails to add geometry to meshcat.

## Reproduction

```python
from pydrake.all import (
    Meshcat,
    Rgba,
    RigidTransform,
    Simulator,
    Sphere,
)
from manipulation.station import LoadScenario, MakeHardwareStation

scenario_data = """
    directives:
        - add_model:
            name: iiwa
            file: package://drake_models/iiwa_description/sdf/iiwa7_no_collision.sdf
        - add_weld:
            parent: world
            child: iiwa::iiwa_link_0
        
    model_drivers:
        iiwa: !IiwaDriver
            control_mode: position_only
    """

scenario = LoadScenario(data=scenario_data)

meshcat = Meshcat()
station = MakeHardwareStation(scenario, meshcat)

# Add red sphere without /drake/illustration prefix -- this doesn't work!
red_ball_path = "/drake/illustration/iiwa/iiwa_link_7/red_ball"
meshcat.SetObject(red_ball_path, Sphere(0.1), rgba=Rgba(1, 0, 0, 1))
meshcat.SetTransform(red_ball_path, RigidTransform([0, 0, 0.1]))

# Add blue sphere without /drake/illustration prefix -- this works!
blue_ball_path = "/drake/abc/iiwa/iiwa_link_7/blue_ball"
meshcat.SetObject(blue_ball_path, Sphere(0.1), rgba=Rgba(0, 0, 1, 1))
meshcat.SetTransform(blue_ball_path, RigidTransform([0, 0, -0.1]))

# Run simulation.
simulator = Simulator(station)
context = simulator.get_mutable_context()
station.GetInputPort("iiwa.position").FixValue(context, [0] * 7)
simulator.set_target_realtime_rate(1.0)
simulator.AdvanceTo(100.0)
```

## Output

I would've expected to see both the red and blue balls, but I only see the blue ball.

[image]

### Version

1.31

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

binary tar.gz download

jwnimmer-tri, 2024-08-05:
The problem was `delete_on_initialization_event=true`, which clears the entire meshcat subtree at initialization-time.  Setting that config option to `false` causes the extra shape(s) to remain intact.

closed 2024-08-05 by jwnimmer-tri

siddancha, 2024-08-05:
Here's an illustration of how setting `delete_on_initialization_event=false` fixes the issue.

### Code

```python
from pydrake.all import (
    Meshcat,
    Rgba,
    RigidTransform,
    Simulator,
    Sphere,
)
from manipulation.station import LoadScenario, MakeHardwareStation

scenario_data = """
    directives:
        - add_model:
            name: iiwa
            file: package://drake_models/iiwa_description/sdf/iiwa7_no_collision.sdf
        - add_weld:
            parent: world
            child: iiwa::iiwa_link_0
        
    visualization:
        delete_on_initialization_event: false
    """

scenario = LoadScenario(data=scenario_data)

meshcat = Meshcat()
station = MakeHardwareStation(scenario, meshcat)

# Add red sphere without /drake/illustration prefix -- this doesn't work!
red_ball_path = "/drake/illustration/iiwa/iiwa_link_7/red_ball"
meshcat.SetObject(red_ball_path, Sphere(0.1), rgba=Rgba(1, 0, 0, 1))
meshcat.SetTransform(red_ball_path, RigidTransform([0, 0, 0.1]))

# Add blue sphere without /drake/illustration prefix -- this works!
blue_ball_path = "/drake/abc/iiwa/iiwa_link_7/blue_ball"
meshcat.SetObject(blue_ball_path, Sphere(0.1), rgba=Rgba(0, 0, 1, 1))
meshcat.SetTransform(blue_ball_path, RigidTransform([0, 0, -0.1]))

# Run simulation.
simulator = Simulator(station)
simulator.set_target_realtime_rate(1.0)
simulator.AdvanceTo(100.0)
```

### Result

https://github.com/user-attachments/assets/28813ff2-ff7f-4959-a964-1931d3153334