RobotLocomotion/drake #21782: AttributeError: module 'drake' has no attribute 'experimental_lcmt_deformable_tri'

XuinloveHuang, 2024-08-06:
[image]
I installed drake through pip: pip install drake but I always get this error when running. Is there any solution? My drake version 1.31.0,my ubuntu is Ubuntu Jammy 22.04

jwnimmer-tri, 2024-08-06:
It works fine for me (see below).  For us to be able to help, you need to provide reproducible instructions for exactly what you did, so that we can follow along and see the same error.

```console
jwnimmer@carcdr:~$ lsb_release -a
...
Description:	Ubuntu 22.04.4 LTS
...
jwnimmer@carcdr:~$ mkdir tmp
jwnimmer@carcdr:~$ cd tmp
jwnimmer@carcdr:~/tmp$ which python3
/usr/bin/python3
jwnimmer@carcdr:~/tmp$ python3 --version
Python 3.10.12
jwnimmer@carcdr:~/tmp$ python3 -m venv env
jwnimmer@carcdr:~/tmp$ env/bin/pip install drake
Collecting drake
  Downloading drake-1.31.0-cp310-cp310-manylinux_2_35_x86_64.whl (79.4 MB)
...
Successfully installed PyYAML-6.0.1 contourpy-1.2.1 cycler-0.12.1 drake-1.31.0 fonttools-4.53.1 kiwisolver-1.4.5 matplotlib-3.9.0 numpy-2.0.1 packaging-24.1 pillow-10.4.0 pydot-3.0.1 pyparsing-3.1.2 python-dateutil-2.9.0.post0 six-1.16.0
jwnimmer@carcdr:~/tmp$ env/bin/python3 -c 'import drake; print(dir(drake))'
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'experimental_lcmt_deformable_tri', 'experimental_lcmt_deformable_tri_mesh_init', 'experimental_lcmt_deformable_tri_mesh_update', 'experimental_lcmt_deformable_tri_meshes_init', 'experimental_lcmt_deformable_tri_meshes_update', 'lcmt_acrobot_u', 'lcmt_acrobot_x', 'lcmt_acrobot_y', 'lcmt_allegro_command', 'lcmt_allegro_status', 'lcmt_call_python', 'lcmt_call_python_data', 'lcmt_contact_results_for_viz', 'lcmt_drake_signal', 'lcmt_force_torque', 'lcmt_header', 'lcmt_hydroelastic_contact_surface_for_viz', 'lcmt_hydroelastic_quadrature_per_point_data_for_viz', 'lcmt_iiwa_command', 'lcmt_iiwa_status', 'lcmt_iiwa_status_telemetry', 'lcmt_image', 'lcmt_image_array', 'lcmt_jaco_command', 'lcmt_jaco_status', 'lcmt_panda_command', 'lcmt_panda_status', 'lcmt_planar_gripper_command', 'lcmt_planar_gripper_finger_command', 'lcmt_planar_gripper_finger_face_assignment', 'lcmt_planar_gripper_finger_face_assignments', 'lcmt_planar_gripper_finger_status', 'lcmt_planar_gripper_status', 'lcmt_planar_manipuland_status', 'lcmt_planar_plant_state', 'lcmt_point', 'lcmt_point_cloud', 'lcmt_point_cloud_field', 'lcmt_point_pair_contact_info_for_viz', 'lcmt_quaternion', 'lcmt_robot_plan', 'lcmt_robot_state', 'lcmt_schunk_wsg_command', 'lcmt_schunk_wsg_status', 'lcmt_scope', 'lcmt_viewer_command', 'lcmt_viewer_draw', 'lcmt_viewer_geometry_data', 'lcmt_viewer_link_data', 'lcmt_viewer_load_robot']

AlexandreAmice, 2024-08-09:
I wanted to note two things that may be causing issues
1.) It looks like you are using a conda environment, [which we don't regularly test with so it is not recommended](https://drake.mit.edu/python_bindings.html)
2.) I frequently see these types of error messages when multiple versions of Drake are installed. Do you perhaps have Drake installed e.g. via apt or your global interpretter?

XuinloveHuang, 2024-08-09:
> I wanted to note two things that may be causing issues 1.) It looks like you are using a conda environment, [which we don't regularly test with so it is not [...]

I have solved the problem, thank you! Indeed, I initially installed it through the source code, and then associated the corresponding site-package in pycharm, but an error occurred when I ran the code to reproduce a paper. Then I create a new conda env,and installed it through pip and forgot to unassociate the site-package installed from the source code, which caused this error.

closed 2024-08-09 by XuinloveHuang