RobotLocomotion/drake #21795: Easy Way to Force a Solver to Use a Single Thread

cohnt, 2024-08-08:
When solving a large number of small mathematical programs in parallel, it's helpful to fix the solver to only using a single thread. It would be great if there were an easy way to set this option, without having to dig into the solver options for whichever solver is being called behind the scenes. Perhaps another CommonSolverOption?

cc @AlexandreAmice

AlexandreAmice, 2024-08-09:
Perhaps an option
```
CommonSolverOptions.kThreadsForSolver, Parrelism::Parallelism()
```
to allow different amounts of threads not just 1 or max?

Either way, I think this might be better suited to someone who understands the implications of changing the number of threads in the solvers? cc @jwnimmer-tri maybe?

This would potentially give a good mechanism for pushing #19119 along.

jwnimmer-tri, 2024-08-26:
I support the idea of common solver option sugar for configuring parallelism.

I would probably keep the `using OptionValue = std::variant<double, int, std::string>;` unchanged, and so the thread count would be an `int` not a `Parallelism`.

cohnt, 2024-08-27:
In that case, presumably we could treat 0 or -1 as "use all threads"?

jwnimmer-tri, 2024-08-27:
Gut feeling, I'd posit that asking the user to `SetOption(CommonSolverOption.kFoo, Parallelism.Max().num_threads())` is better than a magic number people would copy-paste.

jwnimmer-tri, 2024-08-27:
I'd say to match the Parallelism constructor specification:
https://github.com/RobotLocomotion/drake/blob/d8b5c8ff7662c72b8e41ee0e85eecdeff8242fe4/common/parallelism.h#L81-L88

- Throw for less than 1.
- Documented to be a bad idea to set to more than the hardware has available, but will do what you tell it to without any warning.

AlexandreAmice, 2024-08-27:
Two important questions:
1.) Suppose a solver explicitly supports setting threads already (e.g. GurobiSolver supports "Threads"). If a user sets the option both manually using the solver-specific option as well as the common solver option which should we use? I support using the solver-specific option.
2.) Suppose a solver **does not** support setting the number of threads (e.g. Clarabel) what should one do?

jwnimmer-tri, 2024-08-27:
(1) Yes.  Solver-specific options always trump the projection of common options into specific options.

(2) Nothing.  Just like when solvers don't support printing to file, they ignore that option.  This is already documented as "If the solver doesn't support the option, the option is ignored" in solver_options.h.

AlexandreAmice, 2024-09-17:
Closed with #21857

closed 2024-09-17 by AlexandreAmice