RobotLocomotion/drake #21802: MultibodyPlant::SetFreeBodyPose() documentation is misleading

SeanCurtis-TRI, 2024-08-09:
### What happened?

## The Problem

The documentation of `MultibodyPlant::SetFreeBodyPose()` claims: "Sets `context` to store the pose `X_WB` of a given `body` B in the world frame W."

I believe this is an anachronism dating from a time when free bodies were a special concept. Since all free bodies have a joint just like every other body this is no longer true. The definition of "free body" is a body that has an upstream quaternion floating joint. So, I can call `SetFreeBodyPose()` on any body that has an upstream quaternion joint.

However, the free body pose should always be interpreted as `X_PB` (where `P` is the parent, in-board frame and `B` is the body frame). If the parent frame happens to be the world, it is `X_WB` as documented. If, however, `P` is some other frame (like when a floating joint is used to simulate a locking/unlocking between two bodies such as described in [this stackoverflow issue](https://stackoverflow.com/questions/78850709/setfreebodypose-is-not-actually-relative-to-the-world-frame)), then the documentation characterizing the transform as `X_WB` is a lie.

## The Proposal

- At a minimum, we need to update the documentation to reflect the current reality.
- We could also provide a second API that would be `SetFreeBodyPoseInWorld()` that would take `X_WB` as input and compute `X_PB` as appropriate, passing it to `SetFreeBodyPose()`.

SeanCurtis-TRI, 2024-08-09:
#19269 suggests that these APIs will simply go away. So, the question is, is it worth cleaning this up?

I'd propose that unless we have an explicit plan for resolving #19269, cleaning up what is in master so that it's not a lie is worthwhile. (I basically have the branch done, except for the broken API.) And perhaps the solution there is not to fix it, but to simply deprecate it as a lie.

@amcastro-tri @joemasterjohn  @xuchenhan-tri

xuchenhan-tri, 2024-08-12:
My vote would be to fix the documentation so that it reflects the implementation. Then we can deprecate all body-based APIs that we don't think are appropriate all in one go when we fix #19269.

SeanCurtis-TRI, 2024-08-12:
I feel that documentation only is insufficient.

1. In the original issues from stackoverflow, the user was using `pydrake`. Which meant they had the normal usage `plant.SetFreeBodyPose(X_WB=...)`. So, editing doxygen won't stop the python call site from lying. (Again, I have a branch that fixes this).
2. For the API `SetFreeBodyPoseInWorld`, no amount of documentation will make this function correct for any 6-dof body whose parent is not the world. True, we can clarify the documentation saying that it only to a subset of "free bodies", but that ain't great. Perhaps simply best to deprecate that at the same time we update documentation (and python parameter names).

xuchenhan-tri, 2024-08-12:
I see. That makes sense. I skimmed over your branch and I agree that fixing the python call site of `SetFreeBodyPose` + deprecating `SetFreeBodyPoseInWorld` is good for now.

SeanCurtis-TRI, 2024-08-13:
I'd appreciate some eyes on my proposed solution. https://github.com/RobotLocomotion/drake/pull/21808.

Specifically, the two revisions in that PR show two different strategies vis a vis pydrake. If I could get a volunteer to look at them and vote on which would be a better path forward -- one is less disruptive to users but requires more drake code (mostly in test).

Anyone can look that has any interest: look at `plant_py.cc` where I've got the two different approaches applied to a single function in two different revisions. Thanks in advance.

xuchenhan-tri, 2024-08-14:
I'd be down to review the PR if you don't need it right away.

xuchenhan-tri, 2024-08-15:
For some reason, I had the misconception that we have a distinction between free bodies and floating bodies.

The distinction is as Sean proposed in the PR: a body is free if it's inboard joint is a floating joint and a body is floating if it's free and its parent is the world body.

Perhaps we have discussed that idea before but never plumbed through the documentation? In any case, I think #21808 is a nice short-term fix.

sherm1, 2024-08-15:
I think we've always used "free" and "floating" synonymously. If "free" is to be the more general category (i.e. arbitrary parent body) we should have named the two 6dof joints "free joints". But they are both "floating".

sherm1, 2024-08-15:
Terminology proposal to reduce the damage we caused with the existing names:
- a "free body" is one whose inboard joint has 6 dofs with respect to its parent body
- a "floating base body" is a free body whose parent body is World
- refer to the 6dof joints as "6 dof joints" as much as possible rather than calling them "floating" which is a terrible name since they are used for all free bodies
- Use "floating base" wherever possible rather than just "floating"

PR #21808 (2024-08-27): Correct MultibodyPlant free body documentation to reflect current implementations
Much of the free body documentation still assumes that all free bodies have the world as parent. That's not generally true. In fact, we advocate simulating things like suction by putting a floating joint between end effector and manipuland and simply locking the joint when the end effector is "holding" the manipuland. The configuration of the joint is always relative to the inboard frame which, in this case, is not the world.

Furthermore, the API Mbp::SetFreeBodyPoseInWorldFrame() doesn't serve its documented pu [... cut]
closed 2024-08-27 by SeanCurtis-TRI