RobotLocomotion/drake #21819: writing diagram to SVG produces UnicodeEncodeError

felix-h-kong, 2024-08-19:
### What happened?

### History on stackoverflow, MWE
This was reported in [this stackoverflow post](https://stackoverflow.com/questions/78881961/writing-diagram-to-svg-uses-ascii-by-default-not-utf-8/78883286?noredirect=1#comment139086759_78883286). Russ Tedrake suggested I raise this issue after he reproduced this error. An MWE is available here: [MWE](https://github.com/felix-h-kong/drake-1.32-unicode-issue-mwe).

### What's wrong?
I get a UnicodeEncodeError on this line in my script:
```
pydot.graph_from_dot_data(diagram.GetGraphvizString())[0].create_svg("diagram.svg")
```
The error reads:
```
'ascii' codec can't encode character '\U0001faa6' in position 1087: ordinal not in range(128)
  File [redacted], line 114, in make_diagram
    pydot.graph_from_dot_data(diagram.GetGraphvizString())[0].create_svg("diagram.svg")
  File [redacted], line 188, in <module>
    diagram, diagram_context, loggers = make_diagram()
                                        ^^^^^^^^^^^^^^
UnicodeEncodeError: 'ascii' codec can't encode character '\U0001faa6' in position 1087: ordinal not in range(128)
```

### What makes the problem appear/disappear
- In the MWE, the problem in the line above goes away if you use the `MultibodyPlant()` constructor, but returns if you use `AddMultibodySceneGraph()`.
- When using `AddMultibodySceneGraph()`, the problem goes away if you comment out the .vtk collision geometry in the SDF file. 

### Version

1.32.0

### What operating system are you using?

macOS 14 (Sonoma)

### What installation option are you using?

pip install drake

jwnimmer-tri, 2024-08-19:
Thanks for the report.

This is working as intended.  Anytime you read or write a file in Python, you should specify an encoding (typically `utf-8`).  This works:

```py
pydot.graph_from_dot_data(plant.GetGraphvizString())[0].write_svg("test.svg", encoding="utf-8")
```

The default

closed 2024-08-19 by jwnimmer-tri

felix-h-kong, 2024-08-19:
Yep, confirmed that it works on my end. Thanks!