RobotLocomotion/drake #21821: Refactor IRIS variants

RussTedrake, 2024-08-20:
As discussed in [this doc](https://docs.google.com/document/d/12-lTnGxeagizA_-eTk_DM9frPwnRBM-900jhOHBS3J4/edit), we've sprouted a number of variants of the basic IRIS algorithm... they are strewn throughout the codebase and a bit hard to find.

Based on a standup discussion following that document, our current plan is to have a single IRIS entry point which dispatches to the proper variant based on the IrisOptions.

RussTedrake, 2024-08-22:
Goal is to add this to the `planning/iris` subdirectory (not `geometry/optimization`)

cohnt, 2025-08-05:
I would be in favor of closing the issue. We really don't ever recommend using IrisNp at this point, and IrisZo and IrisNp2 have a common interface.

closed 2025-08-13 by RussTedrake