RobotLocomotion/drake #21822: Upgrade IrisInConfigurationSpace to IRIS-NP2

RussTedrake, 2024-08-20:
In our latest paper submission, @cohnt , @rhjiang , and @wernerpe made major improvements to IRIS-NP (the algorithm currently implemented in IrisInConfigurationSpace).  Regions can be computed order(s) of magnitude faster, they tend to be bigger, and tend to have less faces!

We should upgrade our Drake implementation to this latest version of the algorithm.

One question is whether we should make the new code use the CollisionChecker interface (#18830). My understanding is that there was one remaining sticking point, which was discussed in [this slack thread](https://drakedevelopers.slack.com/archives/C0557S260N8/p1719572390231669).

cohnt, 2024-08-20:
Two quetsions from me...

- Re `CollisionChecker`, does that provide sufficient information to solve the counterexample programs? Or would that be given in addition to the plant?
- Do we want to replace IRIS-NP altogether, or does it make sense to keep the old version in for backwards-compatibility (and for replicating previous results)?

RussTedrake, 2024-08-20:
- The goal would be to use CollisionChecker directly, instead of using MultibodyPlant. It is more general.
- I think we should replace IRIS-NP altogether. Git remembers.

cohnt, 2024-08-21:
Extending IRIS-NP to support CollisionChecker seems somewhat orthogonal to upgrading to IRIS-NP2. A lot of the features that CollisionChecker supports need to be taken into account by the counterexample search programs. Negative padding seems particularly complicated to implement. Maybe we can keep these as separate TODOs?

RussTedrake, 2024-08-21:
My understanding is that @rhjiang already has a version that works, except not for the blocker discussed in the slack thread I linked above.  If that's not the case, then i agree we can/should separate them.

rhjiang, 2024-08-21:
The version I have uses CollisionChecker only for querying points in collision.  Not sure I fully understand Tommy's comment, but, the version I have formulates the closest collision program the same way IRIS-NP did, not involving the CollisionChecker.

RussTedrake, 2024-08-22:
In that case, I'm ok deferring https://github.com/RobotLocomotion/drake/issues/18830.  But I guess you construct the SceneGraphCollisionChecker inside the iris call? (the user doesn't have to pass it in?)

RussTedrake, 2024-08-22:
Plan is to _not_ upgrade in place, but to make a new entry point (behind the new common interface from #21821).

PR #23001 (2025-06-13): Implement IRIS-NP2
This adds the minimal functionality for IRIS-NP2 to begin being used. Additional features will be added following this PR, but we're already over 1000 lines. This does _not_ close #21822.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/23001)
closed 2025-06-13 by cohnt