RobotLocomotion/drake #21823: IrisInConfigurationSpace should support floating joints

RussTedrake, 2024-08-20:
IrisInConfigurationSpace currently requires bounded domains for each configuration space variable, except for handling continuous revolute joints in particular cases. It does not currently support any MultibodyPlant with floating base bodies. (This was requested most recently by @lujieyang, @pangtao22, and @hjsuh94.)

This should probably happen after #21822 lands (the upgrade to IRIS-NP2). And based on my f2f with @cohnt , doing this in RPY is not too difficult... the work is mostly about making it easier for the user and/or instructing the user how to do it.

cohnt, 2024-08-20:
Sounds like we need a couple of things:

1. `GetContinuousRevoluteJointIndices` should support `RpyFloatingJoint`. (Right now, it only supports `RevoluteJoint` and `PlanarJoint`.
2. IRIS and its variants should automatically impose the convexity radius limits to continuous revolute joints.
3. IRIS should send a helpful error message if you try to use a `QuaternionFloatingJoint`, suggesting the use of `RpyFloatingJoint` instead.

Was there other functionality that you wanted to have besides these things? If not, I can probably get this stuff done before the big IRIS refactor.

RussTedrake, 2024-08-20:
That seems like the right list to me!

cohnt, 2024-09-10:
Right now, it's actually pretty difficult to get a free object into the scene with a `RpyFloatingJoint` instead of a `QuaternionFloatingJoint`. Maybe this should be a separate issue in its own right, but what do you think about these options?

- A flag in the directives yaml file to add an object with one type or the other (or even a global flag to always use one or the other).
- A config option in the parser (or the plant itself) to use one joint type or the other.

jwnimmer-tri, 2024-09-10:
In terms of a config flag that spans all models, I think that's already filed/tracked as #20943 and is probably not that difficult to add now.  You could ping Sherm in that issue to ask about it.

For something like changing the dmd `AddModel` syntax to support a per-model option for the joint type, that's plausible but should probably be a new issue.

cohnt, 2024-09-11:
@RussTedrake with #21861, we now fully support RpyFloatingJoints in IrisInConfigurationSpace and GcsTrajectoryOptimization. Of course, a config to actually use these types of floating joints (instead of QuaternionFloatingJoint) is still forthcoming. Do you want to close this issue, or keep it open until that functionality is added?

RussTedrake, 2024-09-12:
I think #21861 sufficiently resolved this issue.  Thanks!

closed 2024-09-12 by RussTedrake