RobotLocomotion/drake #21836: rust dependencies spoil debug symbol loading in gdb

rpoyner-tri, 2024-08-21:
### What happened?

* Do a debug build. 
* Try to debug any unit test that has clarabel_solver in its dependency tree (almost anything now that geometry/proximity has the dependency).
* Notice that source line viewing and line numbers in general are not available.

### Version

master circa 1.32.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
rico@PUGET-255560:~/checkout/drake$ bazel test -c dbg //geometry/proximity/...
INFO: Analyzed 730 targets (0 packages loaded, 0 targets configured).
INFO: Found 235 targets and 495 test targets...
INFO: Elapsed time: 0.216s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action

Executed 0 out of 495 tests: 495 tests pass.
rico@PUGET-255560:~/checkout/drake$ gdb --args ./bazel-bin//geometry/proximity/inflate_mesh_test
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bazel-bin//geometry/proximity/inflate_mesh_test...
warning: Missing auto-load script at offset 0 in section .debug_gdb_scripts
of file /home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/geometry/proximity/inflate_mesh_test.
Use `info auto-load python-scripts [REGEXP]' to list them.
(gdb) b inflate_mesh_test.cc:91
No source file named inflate_mesh_test.cc.
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb) quit
```

rpoyner-tri, 2024-08-21:
More detail: using a non-clarabel-infected program still works. Also, hacking out the clarabel dependency also works to restore full debug symbols.

RussTedrake, 2024-08-21:
Nice!  I had noticed this, too, but had not investigated the root cause.

rpoyner-tri, 2024-08-21:
I suspect some cure can be found by reading up about rust debugging. I've seen evidence of rust users having some version of this problem.

rpoyner-tri, 2024-08-24:
Well, here are some completely unsatisfying partial answers.

* get any source file line number symbols at all: add `rustc_flags = ["--emit=obj"],` to the rust static library rule in clarabel_cpp_internal. This results in sources for rust modules and for some drake dependencies, but no drake modules.
* get all source file line number symbols: switch to `mold` linker (https://github.com/rui314/mold). This results in line number symbols from all modules, but (currently) requires a local build of `mold` and some bazel configuration I haven't yet discovered.  I forced the issue by redirecting the symlink at `/usr/bin/ld.gold`, but this is clearly unacceptable.

jwnimmer-tri, 2024-08-27:
FYI workaround: add `--define=NO_CLARABEL=ON` to your bazel command line to opt-out of Clarabel.

sherm1, 2024-09-24:
The NO_CLARABEL workaround doesn't seem to be working for me (on 22.04). The debugger says:
> (No debugging symbols found in multibody/plant/multibody_plant_test)

Command line and debug startup is below. (I got the same behavior using `-c dbg` so tried being more explicit)
```
bazel run --compilation_mode=dbg --run_under=gdb --copt=-g --define=NO_CLARABEL=ON multibody/plant:multibody_plant_test
INFO: Invocation ID: c1f76667-46b2-4d87-87cc-a8a44b4e74f2
WARNING: Build option --copt has changed, discarding analysis cache (this can be expensive, see https://bazel.build/advanced/performance/iteration-speed).
INFO: Analyzed target //multibody/plant:multibody_plant_test (0 packages loaded, 9285 targets configured).
INFO: From Linking multibody/plant/multibody_plant_test:
/usr/bin/ld.gold: warning: bazel-out/k8-dbg/bin/external/fcl_internal/_objs/fcl/mesh_collision_traversal_node.pic.o: top level DIE is not DW_TAG_compile_unit or DW_TAG_type_unit
/usr/bin/ld.gold: warning: bazel-out/k8-dbg/bin/external/fcl_internal/_objs/fcl/kDOP.pic.o: top level DIE is not DW_TAG_compile_unit or DW_TAG_type_unit
/usr/bin/ld.gold: warning: bazel-out/k8-dbg/bin/external/libjpeg_turbo_internal/_objs/jpeg/jddctmgr.pic.o: top level DIE is not DW_TAG_compile_unit or DW_TAG_type_unit
/usr/bin/ld.gold: warning: bazel-out/k8-dbg/bin/external/vtk_internal/_objs/VTK__CommonDataModel/vtkPlanes.pic.o: top level DIE is not DW_TAG_compile_unit or DW_TAG_type_unit
/usr/bin/ld.gold: warning: bazel-out/k8-dbg/bin/external/vtk_internal/_objs/VTK__CommonDataModel/vtkCubicLine.pic.o: top level DIE is not DW_TAG_compile_unit or DW_TAG_type_unit
INFO: Found 1 target...
Target //multibody/plant:multibody_plant_test up-to-date:
  bazel-bin/multibody/plant/multibody_plant_test
INFO: Elapsed time: 201.306s, Critical Path: 168.00s
INFO: 2079 processes: 1 internal, 2078 linux-sandbox.
INFO: Build completed successfully, 2079 total actions
INFO: Running command line: external/bazel_tools/tools/test/test-setup.sh /bin/bash -c '"$@"' /bin/bash gdb multibody/plant/multibody_plant_test
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
Executing tests from //multibody/plant:multibody_plant_test
-----------------------------------------------------------------------------
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.

Reading symbols from multibody/plant/multibody_plant_test...

warning: Could not find DWO CU bazel-out/k8-dbg/bin/multibody/plant/_objs/multibody_plant_test/multibody_plant_test.pic.dwo(0x7f1855f3f292d6c6) referenced by CU at offset 0x0 [in module /home/sherm/.cache/bazel/_bazel_sherm/e6c506c79a0a2d8ce8be77e2bf55a048/execroot/drake/bazel-out/k8-dbg/bin/multibody/plant/multibody_plant_test]
Dwarf Error: unexpected tag 'DW_TAG_skeleton_unit' at offset 0x0 [in module /home/sherm/.cache/bazel/_bazel_sherm/e6c506c79a0a2d8ce8be77e2bf55a048/execroot/drake/bazel-out/k8-dbg/bin/multibody/plant/multibody_plant_test]
(No debugging symbols found in multibody/plant/multibody_plant_test)
```

jwnimmer-tri, 2024-09-24:
That is a different problem, unrelated to Rust.  Doing `bazel build --compilation_mode=dbg --copt=-g --define=NO_CLARABEL=ON multibody/plant:multibody_plant_test` and then `gdb --args bazel-bin/multibody/plant/multibody_plant_test` seems to work fine.  I get stack traces with line numbers & etc.  I suppose when you use `bazel run` to run a test, the paths to the symbol files do not resolve correctly.

rpoyner-tri, 2024-09-30:
Now that #21961 has landed, using replacement linker `mold` should be a usable workaround for debug builds.
```
$ apt install mold  # The version in jammy is good enough
$ bazel test -c dbg --linkopt=-fuse-ld=mold //geometry/proximity:inflate_mesh_test
$ gdb --args ./bazel-bin/geometry/proximity/inflate_mesh_test
(gdb) b inflate_mesh_test.cc:50
(gdb) r
# breakpoint is hit, and `list` command works
```

jwnimmer-tri, 2024-09-30:
In that case, can we automate the fix (and thus close the issue)?  We can easily add `mold` to `setup/ubuntu/source_distribution/packages-*-test-only.txt`.  The question is how to add linkopt to `tools/ubuntu.bazelrc` in some fashion, ideally only for debug builds?

rpoyner-tri, 2024-09-30:
I've been puzzling over the "how to add flags to bazel only for debug" for some time. Still got nothin'. But, I haven't gone dumpster-diving in bazel source code.

jwnimmer-tri, 2024-09-30:
If we're willing to do it in skylark, we could imagine using `select({"//tools/cc_toolchain:debug" ... })` like we do to toggle some tests currently.  If we only need to change test program linking, maybe that's not too bad.

rpoyner-tri, 2024-09-30:
Probably adding some starlark to drake_cc*test and drake_cc*binary would cover it.

rpoyner-tri, 2024-10-02:
A minor impediment to automating `mold` usage: the Jammy package is old enough that builds will fail the exported_symbols_test, but pass with a late-model built-from-source `mold`.

rpoyner-tri, 2025-08-05:
#21961 got reverted for "no more disk space" problems in CI.