RobotLocomotion/drake #21839: IrisInConfigurationSpaceFromCliqueCover should accept sample points and/or edges as an input

RussTedrake, 2024-08-22:
Users should be able to provide more guidance to the coverage... e.g. running the clique cover on points initially obtained via inverse kinematics, imitation data, or planning.

@wernerpe said there were some technical details about the current implementation which make this nontrivial. Opening this issue to track the request.

wernerpe, 2024-08-22:
Yep. The issue is the following: 
We obtain the regions by taking the minimum volume circumscribing ellipsoid of a clique on the visibility graph and use that ellipsoid to seed and run IRIS-NP for a single iteration. If this circumscribing ellipsoid is in collision, then the resulting region may not contain all or any clique vertices. For example, if a user provides a bunch of configurations of a robot arm reaching into a shelf and runs IrisInConfigurationSpaceFromCliqueCover they may get a bunch of regions that contain _no configurations in the shelf_. 

We have a way of getting around this issue by changing the face oritentation for iris if it chops off points of the cluster. We also have a new way of obtaining the sets that eliminates iris completely. We could add the face reorientations directly as an iris feature for now and then move to the new version later? What do you think?

RussTedrake, 2024-08-23:
Thanks Pete. I'd love to get on a whiteboard with you to make sure I understand.