RobotLocomotion/drake #21855: [IRIS] Implement "clique inflation"

RussTedrake, 2024-08-28:
@wernerpe has a variant of IRIS-ZO which uses a distance metric based on point-to-VPolytope instead of the ellipsoidal metric that we typically use in IRIS. One primary feature of this is that it becomes easier to guarantee that the resulting polytope contains all of the points in some initial clique.

This could unblock https://github.com/RobotLocomotion/drake/pull/21838.

wernerpe, 2024-08-28:
For reference: In Iris we use the ellipsoid to measure distance to obstacles ( through the metric induced by the ellipsoid) and for deciding how place faces ( by making them tangent to the expanded ellipsoid).  The idea for clique inflation is simple: replace the ellipsoid with the function that measures distance to the convex hull of all the points we want in our region. This function is easily computable, and automatically guarantees that the resulting regions contain all points, provided that their convex hull is collision free. This means we can guarantee connectivity of regions which means we can inflate PWL paths to corridors, inflate PRMs and cliques on the visibility graph. Below is a sketch describing the idea. A first implementation can be found [here](https://github.com/wernerpe/drake/blob/feature/fastcliqueinflation/planning/iris/fast_clique_inflation.cc).
[image]

wernerpe, 2024-09-25:
See dev branch: https://github.com/wernerpe/drake/tree/devel/cliquecovers